guysoft/FullPageOS #366: unable to login to desktop

nenadlazovic, 2021-02-15:
Issue: unable to login to matchbox desktop using standard pi/raspberry credentials.
Expected: Able to login to desktop using standard pi/raspberry credentials.

Details of what has been tried:
Hi, I made a build of FullPage OS and for a purpose of testing disable starting chromium on boot (simple scripts/run_onepageos to exit right away). As expected Chromium does not start, greeter is shown. Entering username: pi / Password: raspberry gets me back to greeter login dialog. Validated that SSH can login with pi/raspberry credentials.
See few logs at the time of attempting to login.

Any suggestions how to enable login to desktop ?

**/var/log/messages**
```
Feb 15 00:18:28 fullpageos lightdm[2395]: Error getting user list from org.freedesktop.Accounts: GDBus.Error:org.freedesktop.DBus.Error.ServiceUnknown: The name org.freedesktop.Accounts was not provided by any .service files
Feb 15 00:18:30 fullpageos kernel: [  266.943248] broken atomic modeset userspace detected, disabling atomic
Feb 15 00:18:31 fullpageos lightdm[2437]: Error getting user list from org.freedesktop.Accounts: GDBus.Error:org.freedesktop.DBus.Error.ServiceUnknown: The name org.freedesktop.Accounts was not provided by any .service files
```

**/var/log/auth.log**
```
Feb 15 00:15:23 fullpageos lightdm: pam_unix(lightdm-greeter:session): session closed for user lightdm
Feb 15 00:15:23 fullpageos systemd-logind[353]: Removed session c1.
Feb 15 00:15:23 fullpageos lightdm: pam_unix(lightdm:session): session opened for user pi by (uid=0)
Feb 15 00:15:23 fullpageos systemd-logind[353]: New session 6 of user pi.
Feb 15 00:15:24 fullpageos lightdm: pam_unix(lightdm:session): session closed for user pi
Feb 15 00:15:24 fullpageos systemd-logind[353]: Session 6 logged out. Waiting for processes to exit.
Feb 15 00:15:25 fullpageos lightdm: pam_unix(lightdm-greeter:session): session opened for user lightdm by (uid=0)
Feb 15 00:15:25 fullpageos systemd-logind[353]: New session c2 of user lightdm.
Feb 15 00:15:33 fullpageos systemd-logind[353]: Removed session 6.
```

guysoft, 2021-02-15:
It says in the log there is an issue with ``fullapgeos.service`` which does not exist AFAIK in the [GUI module](https://github.com/guysoft/CustomPiOS/tree/devel/src/modules/gui). 

Can you provide: 
``/etc/lightdm/lightdm.conf``
and
``/home/pi/scripts/start_gui``
And if you made any changes to the GUI module?

Also, you can change what is the UI that starts on boot by changing this script:
https://github.com/guysoft/FullPageOS/blob/devel/src/modules/fullpageos/filesystem/home/pi/scripts/start_chromium_browser

Or this config variable to point to a different script:
https://github.com/guysoft/FullPageOS/blob/devel/src/config#L11

Also: might be a keyboard layout issue. Someone had [the same issue here on arch](https://bbs.archlinux.org/viewtopic.php?id=186550) with the same error.

nenadlazovic, 2021-02-16:
Required files attached (had to add .txt extension to be able to attach).
the only change I did was to comment out line from start_gui
```
start_gui
    ...
    #/home/pi/scripts/run_onepageos

```
[start_gui.txt](https://github.com/guysoft/FullPageOS/files/5991199/start_gui.txt)
[lightdm.conf.txt](https://github.com/guysoft/FullPageOS/files/5991200/lightdm.conf.txt)

guysoft, 2021-02-17:
You commented out the line
```
#/home/pi/scripts/run_onepageos
```

That script needs to be blocking. That is, it need to run endlessly. When it terminates you are redirected to the login page.
You need to add instead of it something like this:
https://unix.stackexchange.com/questions/366077/is-there-a-linux-command-that-does-nothing-but-never-exits

You need the gui to run something, otherwise it will just exist to the login screen like you describe.

nenadlazovic, 2021-02-24:
i did as suggested, added
`
    while true; do sleep 600; done
`

to start_gui, instead of invoking /home/pi/scripts/run_onepageos. see below.
No luck. screen is now black with mouse in the middle. no login screen :(

```
root@fullpageos:/home/pi# cat scripts/start_gui
#!/bin/bash
xset s off         # don't activate screensaver
xset -dpms         # disable DPMS (Energy Star) features.
xset s noblank     # don't blank the video device

compton -b -d :0 --backend glx --unredir-if-possible --glx-swap-method buffer-age --glx-no-stencil --paint-on-overlay --vsync opengl
matchbox-window-manager &

#/home/pi/scripts/run_onepageos
while true; do sleep 600; done
```

guysoft, 2021-02-24:
Again, as expected, you started a UI without anything.
I am not sure what you are trying to do.
Raspsberrypi desktop is not shipped on FullPageOS. The idea of FullPageOS is to be a lightweight OS. Otherwise the image would be a few more gigs.

You can install the desktop using the desktop install script in OctoPI:
https://github.com/guysoft/OctoPi/blob/devel/src/modules/octopi/filesystem/home/pi/scripts/install-desktop

Then comment out the line in ``/etc/lightdm/lightdm.conf``:
```
user-session=guisession
```

This uncomment comes from what is done in this line of code which adds it:
https://github.com/guysoft/CustomPiOS/blob/devel/src/modules/gui/end_chroot_script#L11