guysoft/FullPageOS #368: FullPageOS 0.12.0 RC1 status

guysoft, 2021-03-03:
FullPageOS 0.12.0 first release candidate.

Features:
* Chromium Version 86.0.4240.197
* RpiOS 2021-01-11
* Some RaspberryPi 4 B that had issues with blank screens should work now too
* ``disable-pinch`` ``overscroll-history-navigation=0`` flag added (Thanks @jadyndev ) https://github.com/guysoft/FullPageOS/commit/6be19414381522821fc49ddddeb6d1bca129bb06
*  Fix issue from #312, fixes #7 (Thanks @jadyndev )
* Use systemd for vnc #323 (Thanks @pide2000 )
* Url handleing fixes for the startup script #346 (Thanks @Arrow768)
* Checking for "FATAL" in the logs as well #359 (Thanks @rademacs)
Please test and report if there are any problems

## Download
http://unofficialpi.org/Distros/FullPageOS/nightly/2021-03-03_2021-01-11-fullpageos-buster-armhf-lite-0.12.0.zip
md5: [``ea8e93b068f1ff237a6a9e2335979e6d``](http://unofficialpi.org/Distros/FullPageOS/nightly/2021-03-03_2021-01-11-fullpageos-buster-armhf-lite-0.12.0.zip.md5)

## no-acceleration variant

http://unofficialpi.org/Distros/FullPageOS/no-acceleration/nightly/2021-03-03_2021-01-11-fullpageos-no-acceleration-buster-armhf-lite-0.12.0.zip
md5: [``56141bf362997f8aac4d590b421e1670``](http://unofficialpi.org/Distros/FullPageOS/no-acceleration/nightly/2021-03-03_2021-01-11-fullpageos-no-acceleration-buster-armhf-lite-0.12.0.zip.md5)

guysoft, 2021-03-04:
Got a report by @amitdar that vnc is not working.  Might be related to #323 . @pide2000 if you have any inpuwit would be helpful

AidanHak, 2021-03-15:
In regards to #348 and updating to this new release - is there any way to update without flashing the storage device again?

guysoft, 2021-03-15:
@AidanHak There is no way. You could try updating the firmware manually on an old installation, but since Rpi Foundation don't really support it, I can't support this neither. 
This question was also asked here #211

AidanHak, 2021-04-07:
I updated a few Pis to 0.12.0 RC1 and still experience the black screen sometimes from #348

guysoft, 2021-04-13:
I am releasing RC2, because the firmware has been updated.
@AidanHak any chance you could test that too?

AidanHak, 2021-04-13:
Ready when you are! @guysoft

DiomedesDominguez, 2021-04-14:
Is this image based on 32bit or 64bit?

guysoft, 2021-04-14:
@DiomedesDominguez 32bit, 64bit is not supported by rpi foundation . see https://github.com/RPi-Distro/pi-gen/issues/481
Its possible to add support similar to what is done in OctoPi here https://github.com/guysoft/OctoPi/issues/711 , but I ahve no time for it.

Also, you can tell its 32bit because the file has the code ``armhf``. armhf is "arm hard float. And it means its 32bit. The code for 64bit on arm is ``arm64``.

AidanHak, 2021-04-14:
Where can I find RC2? @guysoft

guysoft, 2021-04-14:
@AidanHak I triggered the build, but didn't have time to open an issue for it yet. You can see it in the nightly folder.

guysoft, 2021-04-18:
Not releasing, test out RC2 at #371

closed 2021-04-18 by guysoft