guysoft/FullPageOS #369: Vagrant graphics session

jnath, 2021-03-11:
Hello i search how to start a xsession with vagrant for fullpageos. I would see the resulte of my dev before build image. Have you a solution for that ?

guysoft, 2021-03-13:
If I understand, you want to run the built image locally so you can test it.

You can't use vagrant because vagrant is a tool for virtualization, that's it is, it assumes your CPU runs the same instruction set. However RaspberryPi uses an arm architecture and not intel/amd.

What you can do is use qemu, which is an emulator, not virtualization.

Take a look at:
https://github.com/guysoft/CustomPiOS/wiki/FAQ#can-i-run-my-distro-on-qemu-to-simulate

guysoft, 2021-05-31:
Closing if no reply