guysoft/FullPageOS #370: Screen wont rotate

jesselee04, 2021-04-08:
#### What were you doing?

Trying to get the screen to rotate  90 degrees with added display_rotate=1 or display_rotate=3 to the /boot/config.txt file. 

#### What did you expect to happen?
Screen to rotate
#### What happened instead?
the screen does not rotate
#### Was there an error message displayed? What did it say?
no error message
#### Version of FullPageOS?
0.12.0
[Can be found in /etc/fullpageos_version ALWAYS INCLUDE.]

#### Screenshot(s) showing the problem:

[If applicable. Always include if unsure or reporting UI issues.]

#### If you are building FullPageOS - provide a build.log that is created for the build

guysoft, 2021-04-08:
Duplicate of: #289
Also see #137 , you may need to disable GPU acceleration.

closed 2021-04-08 by guysoft

jesselee04, 2021-04-08:
When i change it to legacy as suggested the boot just stops on " [OK] Started OpenBSD Secure Shell Server ete a swap file.ed" and sits there. never completes boot to gui.

guysoft, 2021-04-09:
Hmm, on that caese it means X crashes.
There might be clues in /var/log/Xorg.0.log

Also found this,  a package may be missing https://raspberrypi.stackexchange.com/questions/100169/xorg-not-working-after-upgrading-to-buster

jesselee04, 2021-04-09:
Something about not being able to load fbturbo module 

    74.107] (==) ModulePath set to "/usr/lib/xorg/modules"
[    74.107] (II) The server relies on udev to provide the list of input devices.
	If no devices become available, reconfigure udev or disable AutoAddDevices.
[    74.107] (II) Loader magic: 0x1fdf80
[    74.107] (II) Module ABI versions:
[    74.107] 	X.Org ANSI C Emulation: 0.4
[    74.107] 	X.Org Video Driver: 24.0
[    74.107] 	X.Org XInput driver : 24.1
[    74.107] 	X.Org Server Extension : 10.0
[    74.109] (++) using VT number 7

[    74.109] (II) systemd-logind: logind integration requires -keeptty and -keeptty was not provided, disabling logind integration
[    74.110] (II) no primary bus or device found
[    74.110] (II) LoadModule: "glx"
[    74.111] (II) Loading /usr/lib/xorg/modules/extensions/libglx.so
[    74.116] (II) Module glx: vendor="X.Org Foundation"
[    74.116] 	compiled for 1.20.4, module version = 1.0.0
[    74.116] 	ABI class: X.Org Server Extension, version 10.0
[    74.116] (II) LoadModule: "fbdev"
[    74.117] (II) Loading /usr/lib/xorg/modules/drivers/fbdev_drv.so
[    74.117] (II) Module fbdev: vendor="X.Org Foundation"
[    74.117] 	compiled for 1.20.3, module version = 0.5.0
[    74.117] 	Module class: X.Org Video Driver
[    74.117] 	ABI class: X.Org Video Driver, version 24.0
[    74.117] (II) LoadModule: "fbturbo"
[    74.118] (WW) Warning, couldn't open module fbturbo
[    74.118] (EE) Failed to load module "fbturbo" (module does not exist, 0)
[    74.118] (II) FBDEV: driver for framebuffer: fbdev
[    74.119] (WW) Falling back to old probe method for fbdev
[    74.119] (II) Loading sub module "fbdevhw"
[    74.119] (II) LoadModule: "fbdevhw"
[    74.119] (II) Loading /usr/lib/xorg/modules/libfbdevhw.so
[    74.120] (II) Module fbdevhw: vendor="X.Org Foundation"
[    74.120] 	compiled for 1.20.4, module version = 0.0.2
[    74.120] 	ABI class: X.Org Video Driver, version 24.0
[    74.120] (EE) open /dev/fb1: No such file or directory
[    74.120] (WW) Falling back to old probe method for fbdev
[    74.120] (II) Loading sub module "fbdevhw"
[    74.120] (II) LoadModule: "fbdevhw"
[    74.121] (II) Loading /usr/lib/xorg/modules/libfbdevhw.so
[    74.121] (II) Module fbdevhw: vendor="X.Org Foundation"
[    74.121] 	compiled for 1.20.4, module version = 0.0.2
[    74.121] 	ABI class: X.Org Video Driver, version 24.0
[    74.121] (EE) open /dev/fb1: No such file or directory
[    74.121] (EE) No devices detected.
[    74.121] (EE)
Fatal server error:
[    74.121] (EE) no screens found(EE)
[    74.121] (EE)
Please consult the The X.Org Foundation support
	 at http://wiki.x.org
 for help.
[    74.121] (EE) Please also check the log file at "/var/log/Xorg.0.log" for additional information.
[    74.121] (EE)
[    74.254] (EE) Server terminated with error (1). Closing log file.
pi@fullpageos:~ $

guysoft, 2021-04-09:
Ok, check out the stack exchange thread I posted.  They mention installing that module.

thfrei, 2022-05-30:
I updated the FAQ: https://github.com/guysoft/FullPageOS/wiki/FAQ#how-to-rotate-the-screen-to-a-vertical-position, please also see: https://github.com/guysoft/FullPageOS/issues/421