guysoft/FullPageOS #371: FullPageOS 0.12.0 RC1 status

guysoft, 2021-04-18:
FullPageOS 0.12.0 second release candidate.

Features:
* Chromium Version 88.0.4324.187
* RpiOS 2021-03-04
* Some RaspberryPi 4 B that had issues with blank screens should work now too Fixes #348
* ``disable-pinch`` ``overscroll-history-navigation=0`` flag added (Thanks @jadyndev ) https://github.com/guysoft/FullPageOS/commit/6be19414381522821fc49ddddeb6d1bca129bb06
*  Fix issue from #312, fixes #7 (Thanks @jadyndev )
* Use systemd for vnc #323 (Thanks @pide2000 )
* Url handleing fixes for the startup script #346 (Thanks @Arrow768)
* Checking for "FATAL" in the logs as well #359 (Thanks @rademacs)
Please test and report if there are any problems

Changes from RC1 - upgraded to 2021-03-04, because the firmware in it should make new hardware work, some Rpi4 mostly.
Chromium version updated.

## Download
http://unofficialpi.org/Distros/FullPageOS/nightly/2021-03-03_2021-01-11-fullpageos-buster-armhf-lite-0.12.0.zip
md5: [``ea8e93b068f1ff237a6a9e2335979e6d``](http://unofficialpi.org/Distros/FullPageOS/nightly/2021-03-03_2021-01-11-fullpageos-buster-armhf-lite-0.12.0.zip.md5)

## no-acceleration variant

http://unofficialpi.org/Distros/FullPageOS/no-acceleration/nightly/2021-04-14_2021-03-04-fullpageos-no-acceleration-buster-armhf-lite-0.12.0.zip

md5: [``e6a4116d83918101e400a69269e6858e``](httphttp://unofficialpi.org/Distros/FullPageOS/no-acceleration/nightly/2021-04-14_2021-03-04-fullpageos-no-acceleration-buster-armhf-lite-0.12.0.zip.md5)

## Known issues:
* Google Chrome 80 introduced SameSite cookie enforcement in February 2020 and some sites using FullPageDashboard require disabling SameSite cookie enforcement in order to login. [Here is how to do it](https://support.siteimprove.com/hc/en-gb/articles/360007364778-Turning-off-Google-Chrome-SameSite-Cookie-Enforcement).

guysoft, 2021-04-26:
Its been a week with no replies, so planning to release

AidanHak, 2021-05-07:
Testing it out now. RC1 is a lot better. Fingers crossed RC2 resolves the issue completely.

jadyndev, 2021-05-15:
I found an issue when using a raspberry pi 3: I can't get the display to work, I'm currently investigating why.
The raspberry boots as usual but when it's time for the splash screen the display isn't receiving anything. Interesting is that there are black bars around the image, regardless of the configuration in `/boot/config.txt`

jadyndev, 2021-05-15:
Okay, found it:
`boot/config.txt`:
```
dtoverlay=vc4-kms-v3d
```
This line seems to break it, it works just fine when it's removed / commented out.

guysoft, 2021-05-27:
Tested, works, releasing

closed 2026-10-07 by guysoft