guysoft/FullPageOS #376: Is it possible to show 2 screens at the same time with splitscreen?

Foxmaner, 2021-06-29:
#### What were you doing?
I try to find a solution for my problem and found this repository. I want a simple OS that shows 2 seperate websites on the same screen (Each website should take half the screen). Would this be possible with FullPageOS?

guysoft, 2021-06-29:
No idea.
I can point you to the name of the tool you are looking for. You need a "window manager". And one that is programmable.

dennisrahmen, 2021-08-17:
Since you want to display two websites you could just create a new "multi_window.html" and show both pages as iframes.
**Keep in mind that the pages you are trying to display could have disabled iframes.**

Create a new file, for example "multi_window.html", in "/var/www/html/" with the following content:
```
<html>
    <body style="margin: 0px;">
        <iframe style="float: left; width: 50%; height: 100vh; border: none;" src="[URL]"></iframe>
        <iframe style="float: right; width: 50%; height: 100vh; border: none;" src="[URL]"></iframe>
    </body>
</html>
```
Replace the "[URL]" with your websites.

Then you have to edit '/boot/fullpageos.txt', to point to your HTML file:
`http://localhost/multi_window.html`

Norfeldt, 2022-02-26:
thanks for the suggestion @dennisrahmen 

Is there any other alternatives than iframes, since as you say some sites will not allow it to be within an iframe.

dennisrahmen, 2022-02-26:
@Norfeldt Yes! Just found one I did not thought about earlier.

Since the browser is enforcing those policies you can tweak you browser to allow them. This also applies if you want to access an API that blocks localhost by CORS policy for example.

In your case I would suggest installing the following extension while using my example above.
https://chrome.google.com/webstore/detail/ignore-x-frame-headers/gleekbfjekiniecknbkamfmkohkpodhe/related

One of the function keys will open a new chrome tab where you have access to the options and the URL, so just install the extension like normal and you are got to go.

Norfeldt, 2022-02-27:
Thanks for the suggestion @dennisrahmen 

I just gave it a try on my laptop and it does not seem to work for me.

[image]

I'm attempting to split-screen a non-public google calendar. (Viewing it by just the URL works fine, just need to sign in to view it)

dennisrahmen, 2022-02-27:
@Norfeldt I would be surprised if this is possible.
Google will try everything they can to avoid people iframing their login page.

So I am out of ideas here.

KyleMoran138, 2022-07-11:
Stumbled here. I'm looking for a solution also.

dennisrahmen, 2023-05-03:
@Norfeldt @KyleMoran138 I found this extension, it supposedly splits the browser window into multiple parts.

https://chrome.google.com/webstore/detail/split-screen-for-google-c/dnollkdkikklpdganoecjcmmlddbennb

Maybe this works for you, you can install it just like the other extension that I previously mentioned here.

Norfeldt, 2023-05-04:
@dennisrahmen thanks for sharing

(FYI edge has an experimental feature that allows "split screen" `edge://flags/#edge-split-screen`)

dennisrahmen, 2023-05-04:
Problem is that FullPageOS does not ship with Edge.

Norfeldt, 2023-05-04:
Yes, you are right. It's just that is also using Chromium so Imagine that it could be a native feature at some point