guysoft/FullPageOS #377: Stuck in Authentication Loop

Dr-Juice, 2021-07-20:
#### What were you doing?

attempting to log into a local served website using username and password

#### What did you expect to happen?

authenticate to website

#### What happened instead?

stuck in a loop back to the login page.  No error message.

#### Was there an error message displayed? What did it say?

No

#### Version of FullPageOS?

2021-04-14_2021-03-04-fullpageos-buster-armhf-lite-0.12.0

#### Screenshot(s) showing the problem:

No screenshot, although authentication works fine if I take chromium out of Kiosk mode and log in using the full browser in FullPageOS

guysoft, 2021-07-20:
Try this:
1. press ctrl+t to open a new tab
2. Authenticate on the new tab
3. Try to open the result url in FullPageOS.

It might be that the website, which I can't see therefore have to guess, is doing authentication wrong. I have had several logging in and working fine on FullPageOS. It might be that once you log in the url changes, or it handles cookies wrong.

Dr-Juice, 2021-07-20:
Thanks Guy.

I’ll give that a try.  I agree it’s hard to explain the issue properly.

Regards
Tony Milazzo

[CAUTION: EXTERNAL EMAIL]

t-knapp, 2023-03-29:
@Dr-Juice Does the server set a cookie with session information and forwards (HTTP 30X) to the "real" URL?