guysoft/FullPageOS #378: Emoji Support

SimonIT, 2021-08-03:
#### What were you doing?

I'm displaying a dashboard with some emojis on it

#### What did you expect to happen?

The dashboard with emojis are displaced

#### What happened instead?

The emojis are not displayed but the rectangles for a not found symbol are displayed.
My suggestion is to install [noto](https://www.google.com/get/noto/) for the widest support.

#### Was there an error message displayed? What did it say?

#### Version of FullPageOS?

0.12.0

#### Screenshot(s) showing the problem:

[image]

#### If you are building FullPageOS - provide a build.log that is created for the build

guysoft, 2021-08-03:
This is something worth adding. Since emojis are becoming ubiquitous and are part of Unicode.
I tried looking up ways to ship this RaspsberryPi and it seems all suggestions on how to do that are being bashed (tastelessly IMHO) on the Rpi fourms.
Example:
https://www.raspberrypi.org/forums/viewtopic.php?t=253484
And:
https://www.raspberrypi.org/forums/viewtopic.php?t=287942

It seems the only way suggested I can find is to download manually the fonts and place them in the /home/pi/.fonts folder.
Something along the lines of:

```
wget --directory-prefix ~/.fonts https://github.com/googlefonts/noto-emoji/raw/master/fonts/NotoColorEmoji.ttf
```

If you could test if running this command solves your problem, it would help because I could ship that.

SimonIT, 2021-08-09:
Yes, that worked perfectly!

commit a67af7e8 (2021-08-09): Add emoji support, fixes #378
closed 2021-08-09 by guysoft