guysoft/FullPageOS #384: Can not establish WPA2-enterprise secured Wifi-network connection

becci00, 2021-09-23:
#### What were you doing?

Trying to establish a WPA2-enterprise secured wifi-network connection using the file _/boot/fullpageos-wpa-supplicant.conf_ on a Raspberry Pi 3

#### What did you expect to happen?

To establish an internet connection via wifi

#### What happened instead?

The connection seems to get established and is visible via iwconfig. Authentication via our RADIUS-Server seems to work, but the device does not receive an IP-address and thus has no internetconnection.
I tried to connect via WPA-PSK as well and this works fine. 

#### Was there an error message displayed? What did it say?

no

#### Version of FullPageOS?

0.12.0

#### Screenshot(s) showing the problem:

This is what the seemingly established connection looks like:
[image]

This is my **WPA2-enterprise configuration in _/boot/fullpageos-wpa-supplicant.conf_ which does not work**:

```
network={
ssid=”ourNetwork”
pairwise=CCMP TKIP
group=CCMP TKIP
scan_ssid=1
key_mgmt=WPA-EAP
identity=”myIdentity”
password=”myPassword”
eap=PEAP
phase1=”peapver=0”
phase2=”MSCHAPV2”
}
```
For connecting with our Notebooks we normally use a certificate. I copied it into _/boot/_ and included it by adding the following line to the network definition:

`ca_cert="/boot/ourCertificate.pem"`

I tried to run this configuration with and without the certificate. It did not seem to matter whether I included it or not.

This is **the WPA-PSK configuration in _/boot/fullpageos-wpa-supplicant.conf_ that works just fine**:

```
Network={
ssid=”our2ndNetwork”
key_mgmt=WPA-PSK
psk=”myPassKey”
}
```

I tried to add a **definition of the wlan0 interface in _/etc/network/interfaces_**, since this was suggested in several online resources:

```
iface lo inet loopback
allow-hotplug wlan0
iface wlan0 inet dhcp
                pre-up wpa_supplicant -B Dwext -I wlan0 -c/boot/fullpageos-wpa-supplicant.conf
                post-down killall -q wpa_supplicant
```

I tried this version as well:

```
auto lo
iface lo inet loopback

auto wlan0
iface wlan0 inet dhcp
                wpa-conf /boot/fullpageos-wpa-supplicant.conf
```

Both of these versions seem to change nothing. Any help is highly appreciated!

thedumbterminal, 2022-03-05:
Have you tried following the setup for rasbian, such as : https://turunen.ee/setting-up-radius-on-a-raspberry-pi/

becci00, 2022-03-08:
Dear stranger, Thank you for the suggestion. I'm afraid the provided link does not help with the issue though. We already do have a radius server in our network and there are multiple Linux-clients which can connect and work just fine. The problem is that the raspberry Pi can not join as a client via WPA2. In that regard, I don't know how it would help to set up our client as a second radius server. Thanks anyway, any further help is appreciated.