guysoft/FullPageOS #389: Motion sensor support: working solution!

TheShanMan, 2021-11-22:
Not a feature request per se so feel free to close, but perhaps it's something worth adding support to if you see this as useful enough to a large enough portion of your users. Otherwise, maybe this is worth sharing in case a user comes here looking for something similar and finds this as a closed issue :).

I use FullPageOS for displaying a security camera stream and wanted to set up a motion sensor to turn off the monitor when no one is in the room. I used https://github.com/bogenpirat/rpi-display-sleep-manager/ to implement it although it took quite a bit of figuring out how to make it work right with FullPageOS.

To run on startup, I had to add this to /home/pi/scripts/start_gui (other methods such as the one mentioned by the author of that script didn't work... I eventually found this solution in a closed FullPageOS issue):
`python3 /home/pi/monitor.py > /tmp/monitor.log 2>&1 &`

The script provided by that project mostly worked as long as you don't try to make use of monitor-helper and _only_ use monitor.py. I also had some trouble with logging so I disabled it in the script and captured logging via the start_gui command above. I also increased the polling frequency to 0.5s and the "on" time to 5 minutes (300s). Finally, the script waits to sense motion on initialization and doesn't really proceed until it first senses it. I altered it to not care if motion is sensed on startup and instead just act as though motion _is_ sensed, such that the monitor will be on for 5 minutes minimum when the script starts, even if no one is present/moving. The script doesn't paste very well here but it's probably better to point back to the original author's work anyway, as I did above. If the description of my tweaks aren't enough, feel free to ping me for the changes I made.

I tweaked the pots on my motion sensor for maximum sensitivity and for the signal to remain active for a second or two, rather than whatever the minimum is. And FWIW I'm using an rpi 3b+.

This is working so well for me. Thank you for this project and I hope someone else finds this motion sensor solution useful.

guysoft, 2021-12-09:
Sounds like a good base for a distro using https://github.com/OctoPrint/CustoPiZer
Which @foosel built just for the case of tweaking something on top of CustomPiOS images

guysoft, 2021-12-09:
Reopen if you need help on building your own distro, but I don't plan to maintain something like that.

closed 2021-12-09 by guysoft