guysoft/FullPageOS #391: Question: reloading / updating running chromium with updated url from fullpageos.txt

shaunhurley, 2021-11-25:
Hey Team,  apologies if this one has been answered, been scratching my head for a while.

Is there anyway from the command line (actually looking to execute remotely via ssh) to have the running instance of chromium reload the page specified in fullpageos.txt without restarting the pi / fullpageos completely?  As in, if I modify the /boot/fullpageos.txt (remotely, via ssh in this case)?  I can do a full restart with a shutdown -r now but it just seems excessive :)

Tried the refresh and safe_refresh scripts, but they reload the current page only, so that doesn't help.

Many thanks!

Shaun Hurley

guysoft, 2021-11-25:
Yes there is.
There are two kinds even, 
refresh, and safe refresh.
Safe refresh checks if there is the internet connection is up and only then refreshes.

The scripts are located at
[``/home/pi/scripts/refresh``](https://github.com/guysoft/FullPageOS/blob/devel/src/modules/fullpageos/filesystem/home/pi/scripts/refresh)

and:
[``/home/pi/scripts/safe_refresh``](https://github.com/guysoft/FullPageOS/blob/devel/src/modules/fullpageos/filesystem/home/pi/scripts/safe_refresh)

Also, I am one person who maintain FullPageOS. So not really a team.

shaunhurley, 2021-11-25:
Hi, thanks for the quick response! And if you are managing this solo, along with all your other projects, then that's even more impressive - FullpageOS is a great system!

I've tried both refresh and safe_refresh, however they appear to only refresh the currently loaded page.  Is there anyway to reload from /boot/fullpageos.txt (assuming it has been updated in the background)

I'm thinking windowclose and/or windowkill from xdotool then do a start fullpageos...

Thanks again!

Shaun

guysoft, 2021-11-28:
Cool, we could add a script for this in the ``scripts`` folder if you think that is helpful.

shaunhurley, 2021-11-28:
Sure, could be useful.  I already have menu options (I have a central pi that is the "controller", running a simple web interface via lighttpd, that manages the remainder of devices connected to tvs) that can invoke a couple of the other scripts (refresh, etc. - no need to go for the hard kill if we don't have to :) so that would work just as well, presumably it might be helpful for some of your other users?

It's a simple patch, do you want me to generate a pull request for it, or throw it in yourself?

guysoft, 2021-12-09:
Naa, did it myself this time, if you want it in a different name of something you can PR that

commit 0fbbe61c (2021-12-09): Fixes #391
closed 2021-12-09 by guysoft