guysoft/FullPageOS #393: http://unofficialpi.org is just a spam page

jayjay-sweden, 2021-12-06:
The build links don´t work. http://unofficialpi.org is only showing ads.

guysoft, 2021-12-06:
Strange I dont see any ads. And the urls load fibe here:
http://unofficialpi.org/Distros/

Can you try another machine?

guysoft, 2021-12-09:
Closing if no reply

jayjay-sweden, 2021-12-09:
I´m sorry for not replying. Waited to check on the same computer...
Well the URL you gave as example works. 
http://unofficialpi.org/Distros/FullPageOS/2019-12-01_2019-09-26-fullpageos-buster-lite-0.11.0.zip and links to some other files does not work.  It only shows the same "ad page" as http://unofficialpi.org/ does.
[image]

guysoft, 2021-12-09:
Its not what I am getting here.
I suspect you have some mallware installed on your machine, that hijacks white empty webpages.

1. the url you provided is an invalid url:
http://unofficialpi.org/Distros/FullPageOS/2019-12-01_2019-09-26-fullpageos-buster-lite-0.11.0.zip
Is not old and has been moved to:
http://unofficialpi.org/Distros/FullPageOS/old_versions/2019-12-01_2019-09-26-fullpageos-buster-lite-0.11.0.zip

*Please dont use it and upgrade*

2.
Can you give a curl output of the page?
I tried it from multiple servers around the world, and this is what I get:

```
$ curl http://unofficialpi.org/
<!DOCTYPE html>
<html>
 <head>
  <meta charset="utf-8">
  <style type="text/css">
   html, body, #partner, iframe {
                height:100%;
                width:100%;
                margin:0;
                padding:0;
                border:0;
                outline:0;
                font-size:100%;
                vertical-align:baseline;
                background:transparent;
            }
            body {
                overflow:hidden;
            }
  </style>
  <meta content="NOW" name="expires">
  <meta content="index, follow, all" name="GOOGLEBOT">
  <meta content="index, follow, all" name="robots">
  <!-- Following Meta-Tag fixes scaling-issues on mobile devices -->
  <meta content="width=device-width; initial-scale=1.0; maximum-scale=1.0; user-scalable=0;" name="viewport">
 </head>
 <body>
  <div id="partner">
  </div>
  <script type="text/javascript">
   document.write(
                    '<script type="text/javascript" language="JavaScript"'
                            + 'src="//sedoparking.com/frmpark/'
                            + window.location.host + '/'
                            + 'IONOSParkingUS'
                            + '/park.js">'
                    + '<\/script>'
            );
  </script>
 </body>
</html>
```

jayjay-sweden, 2021-12-09:
Actually its the script in your code from sedoparking thats is responsible for displaying the ads...
(Shouldn't really be there i think, unless the domain is really parked, which i guess it isn't)
If i block that script i get a white / blank page instead. I guess that is what you see. 

Anyway. The links i have pasted here comes from the page https://github.com/guysoft/FullPageOS/releases
So if people should use https you should probably update them 😉

Guess its the links to files that have been moved that shows ads.

guysoft, 2021-12-10:
Hey, thanks for reporting and looking in to it. It seems all my OpenWrt and
ad blockers keep me from seeing the ads. Will try investigate if something
on the server was compromised.

>
There is also octopi.gnethomelinux.com that mirror that server. Need to
check if it mirrored the ads.

guysoft, 2021-12-11:
Ok, I did some investigation, and thanks to @foosel that also helped me test what is going on.

First, there workaround is that you can use the second mirror we have: https://octopi.gnethomelinux.com/Distros/FullPageOS/

It has the correct behavior. It seems like something is strange with forbidden or non-existent pages. They redirect to "sedoparking" something which I will not post here.

Will make contact with Doc who donated to server to find out why this is happening.
Currently non-empty pages seem to work fine.

jayjay-sweden, 2021-12-11:
Yes I already got the files i needed. ( Thanks for those btw.) Was more that l was surprised when I clicked one of the links and ended up on an ad page.

Yup that second mirror seems to work as expected.

guysoft, 2021-12-13:
Update from Doc who donated the server, apparently the service provider just added a redirection to their own website, we can change it to something else. So no compromises or something like that.

guysoft, 2022-02-05:
Fixed

closed 2022-02-05 by guysoft