guysoft/FullPageOS #394: How to add a python script to automatically start at bootup?

AlmJoDla, 2021-12-11:
Hi!
I successfully setup fullpageos which will deliver a website to a digital signage tv. 
I also created a python script which sends Standby/On commands to the tv via HDMI-CEC. 
The script is already tested and working fine on a different raspberry pi. 

Now I would like to install the python script on fullpageos so that it will start on bootup in the background. 

I managed to leave fullpageos to command line and only found the folder scripts. 

Is it possible to put my script here? How can configure it to be executed? 
Any other methods of executing a python script on bootup on fullpageos?

Any hints are greatly appreciated.

guysoft, 2021-12-11:
Hey,
It depends how you want it to auto-start. The ``home/pi/scripts`` folder holds lots of useful scripts you can use. You could place it there. 
If you want to share your script it might be possible to incorporate it in to FullPageOS if its generic enough.

AlmJoDla, 2021-12-11:
Hi Guysoft!
I would like to place it into scripts but I have no idea how to make it autostart. Your other proposal is to just start the script from one of the other scripts or add them there right?

The script does not depend on the GUI. It just does some basic http request to my automation system which delivers a status. Depending on the status I send Standby/On to the monitor via HDMI-CEC (using cec-utils)

I will try adding 
os.system("myscript.py") 
to it from start_chromium_browser and see what happens :-)

guysoft, 2021-12-11:
``start_chromium_browser`` is using the bassh shell interpreter, not python.You might need to add ``&`` so it forks the process.

If you share the script it would help to understand what is going on.

AlmJoDla, 2021-12-11:
#!/usr/bin/python
import time
import requests
import os
import re

#HDMI-CEC Code uses python cec tool to send hmdi signal to tv
#time.sleep(10)

print("Waiting 10 Seconds after bootup till LAN is available...")
time.sleep(10)

try:

	while True:
		time.sleep(5)

		print("Press Ctrl-C to terminate script")			

		# Make connection, print response and close the connection
		r = requests.get('http://IP/dev/sps/io/BeleuchtungAufenthalt/State', auth=('username', 'password'))
		#print(r.text)
		r.close()

		#Process the result
		match = re.search("value=\"\d", r.text)
		buffer = match.group(0)
		state = buffer[7:8]

		#Depending on the result send Standby or  on via HDMI-CEC

		if state=="1":
			print("Jemand ist da - State 1")
			#os.system("echo 'standby 0.0.0.0' | cec-client -s -d 1")

		else:
			print("Niemand ist da - State 0")
			#os.system("echo 'on 0.0.0.0' | cec-client -s -d 1")
			print(state)

except KeyboardInterrupt:
	print("Script terminated")
	pass

AlmJoDla, 2021-12-11:
So I added this line to start_chromium_browser
`sudo python /home/pi/scripts/hdmi.py &`

I dont know how to run start_chromium_browser from console. To see if the above command is executed I just added a single line of code to hdmi.py where i delete a dummy file. Therefore I can tell if it was executed during startup. It doesnt work. If i execute the command from the console it works fine. 

Sorry for my crude technique here but I am a bit of a beginner on linux based systems.

AlmJoDla, 2021-12-11:
I did a lot of trial and error with no real success. 
I created a new bash file called hdmi

`#!/bin/bash`
`python /home/pi/scripts/hdmi.py`

I also added a line into run_onepageos
`/home/pi/scripts/hdmi` right after `#!/bin/bash`

When I execute run_onepageos from console hdmi executes hdmi.py and deletes me testfile. But when I reboot the system to have my python script executed it doesnt do it.

guysoft, 2021-12-12:
``start_chromium_browser`` starts the chromium browser, if you reboot the device its executed.

That script only does https requests and does not actually do anything because the commands are commented out.

AlmJoDla, 2021-12-12:
Hi Guysoft!
Thank you for your help! I am aware that this is a very specific problem and is not really relevant for FullPageOS. I learned a lot already and will figure it out somehow. The critical HDMI-CEC commands were disabled on purpose to not shut of my monitor all the time ;-) 
Cheers
Johannes

peteruithoven, 2024-12-21:
I've added a relevant guide which uses a systemd service to do the same: 
https://github.com/guysoft/FullPageOS/wiki/Control-TV-over-HDMI