guysoft/FullPageOS #395: Cable or wifi in intranet

davisukin, 2022-01-09:
Hello,
I am newbie to raspberry.
I am trying to install a FullpageOS on a raspberry within an intranet, which uses static ip.
I have all of info: ip, wateway. dns, but I don't know where / how to put them.
The raspberry can be connected to inet via cable or wifi.
The tests outside the intranet (in my house) have been satisfactory since I use wifi with dhcp configured in "fullpageos-wpa-supplicanty.txt" and the page to be viewed is a public website.

Regards.

#### Version of FullPageOS?

2021-03-04-fullpageos-buster-armhf-lite-0.12.0

guysoft, 2022-01-09:
What is your question?

davisukin, 2022-01-09:
How to configure a static ip.
Can i do in the fullpageos-wpa-supplicanty.txt for the wifi?
How cain i do for cable'

guysoft, 2022-01-09:
Duplicate of #222
See https://community.octoprint.org/t/static-ip-adress-for-wifi-and-cable-is-done/5600/2
You can't set static IP in wpa-supplicant because wpa-supplicant is part of the wireless encryption system and not part of the IP handling system. It goes in ``/etc/dhcpcd.conf``

closed 2022-01-09 by guysoft