guysoft/FullPageOS #396: Emoji Support

havenonick, 2022-01-28:
#### What were you doing?

Trying to display a webpage with Emojis on it.

#### What did you expect to happen?

Show the Emojis

#### What happened instead?

Funny symbols showed up.

#### Was there an error message displayed? What did it say?

#### Version of FullPageOS?

Newest from today.

#### Screenshot(s) showing the problem:

#### If you are building FullPageOS - provide a build.log that is created for the build

guysoft, 2022-01-29:
Duplicate of #378

guysoft, 2022-01-29:
@havenonick There is a fix in master that is not released, If you can try the latest nightly, it should solve the issue (and it would be great if you can report back to #378

closed 2022-01-29 by guysoft