guysoft/FullPageOS #397: How to increase DPI?

Alternativend, 2022-01-28:
#### What were you doing?

We are having trouble in our school with an 4k display which is too small to read it. 

#### What did you expect to happen?

I would like to increase DPI to a fix value. Is this possible somewhere?

Thanks!

guysoft, 2022-01-29:
Hey @Alternativend cool to know FullPageOS found its way to a school. If you want to share what you are doing that would be great!

FullPageOS uses lightdm window manager to display everything, so in order to tell it to display fonts in a higher DPI you need to change the settting in ``/etc/lightdm/lightdm.conf``:
In ``/etc/lightdm/lightdm.conf``, you can override the xserver-command parameter (which usually defaults to simply "X"). Here the dpi setting:

You need to uncomment it in the ``[Seat:*]`` section.
```
xserver-command=X -dpi 75
```

Source:
https://askubuntu.com/questions/72486/can-i-pass-a-dpi-flag-to-lightdm-to-set-the-proper-dpi-for-x

Also took a look at the lightdm.conf of current stable to see how it looks.

Alternativend, 2022-01-30:
Well that did not do the trick for me. 

I ended up editing 

/home/pi/scripts/startchromium_browser

to 

`chromium-browser --kiosk --touch-events=enabled --disable-pinch --noerrdialogs --disable-session-crashed-bubble --app=$(head -n 1 /boot/fullpageos.txt | sed -e "s/{serial}/${SERIAL}/g") **--force-device-scale-factor=1.79**`

guysoft, 2022-01-31:
Good to know that solved it. Pics would be great, either here or if you want to post me to guysoft at gmail dot com.

Alternativend, 2022-01-31:
[image]

guysoft, 2022-02-08:
Cool! Thanks!