guysoft/FullPageOS #398: New version - changes?

TiWe1998, 2022-01-31:
Good day :)

After a long time there is a Nightly Build again. Is there a place where I can see the changes? Or can you tell me what has changed or what will be implemented in the future?

Thanks in advance!

BR

guysoft, 2022-01-31:
The commit logs here and in CustomPiOS repo.
I planning to make am RC1 for OctoPi today. Will make one for FullPageOS and compile a list of everything in one place.

TiWe1998, 2022-02-04:
Hello guysoft,

Is it correct that now over the FullpageOS there is a bar with "FullPageOS Dashboard - Website" in blue? Can you hide these? 

And the splash screen doesn't work (anymore). I used the 02/02/22 version.

I also had the problem that the dashboard was only displayed after the 3-4 restart. This was also reported by another user.

BR

guysoft, 2022-02-04:
Hey,
1. Ill have a look at the splash, perhaps bullseye changed something. If anyone has a  video it would help.
2. RaspberryPi OS reboots multiple times to enlarge the filesystem, this is how they manage the first time boot. Its not something related to FullPageOS

mrosu-sp, 2022-06-14:
> I once made you a video of the start. At the end you can see again which bar I mean. Why is this there now? :)
> 
>  pxl-20220204-121547673_24UNz5bl.1.mp4

I have the same issue with version 2022-01-28.

mrosu-sp, 2022-06-16:
I applied the following changes to /boot.config.txt and the issue has gone away:

disable_overscan=1
hdmi_force_hotplug=1
hdmi_group=1
hdmi_mode=16
#dtoverlay=vc4-fkms-v3d