guysoft/FullPageOS #400: Blank screen - Chromium error

AidanHak, 2022-02-01:
#### What were you doing?

Updated via `sudo apt-get update` and `sudo apt-get upgrade`

The Pi loads up and then goes to a blank (black) screen.

#### What did you expect to happen?

I expected the page to load based on the URL I put into `/boot/fullpageos.txt`

#### What happened instead?

The Pi loads up and then goes to a blank (black) screen.

#### Was there an error message displayed? What did it say?

```
pi@OV-DIR-B:~/scripts $ chromium-browser
dpkg-query: no packages found matching bluealsa
[0201/105336.046106:ERROR:elf_dynamic_array_reader.h(61)] tag not found
Trace/breakpoint trap
```

#### Version of FullPageOS?

`2022-01-26_2021-10-30-fullpageos-default-bullseye-armhf-lite-0.13.0` released on **2022-01-25 18:31**

[Can be found in /etc/fullpageos_version ALWAYS INCLUDE.]

**0.13.0**

AidanHak, 2022-02-01:
Here is a list of my configuration steps:

`sudo apt-get update`

`sudo raspi-config`
_Set password, hostname
Enable VNC
Localisation
Expand Filesystem_

`sudo reboot`

`sudo apt-get update`

`sudo apt-get upgrade`

`sudo apt-get install unclutter`

`sudo nano /home/pi/scripts/start_chromium_browser`
_Add `--disable-component-update` option_

`sudo touch /etc/chromium-browser/customizations/01-disable-update-check;echo CHROMIUM_FLAGS=\"\$\{CHROMIUM_FLAGS\} --check-for-update-interval=31536000\" | sudo tee /etc/chromium-browser/customizations/01-disable-update-check`

`sudo nano /etc/dphys-swapfile`
_Edit the CONF_SWAPSIZE value to be 1000 or greater_
`sudo /etc/init.d/dphys-swapfile stop`
`sudo /etc/init.d/dphys-swapfile start`

`sudo reboot`

guysoft, 2022-02-02:
What version of FullPageOS are you using? Have you tried using the nightly build?

AidanHak, 2022-02-04:
> What version of FullPageOS are you using? Have you tried using the nightly build?

As I mentioned, I'm on the newest nightly build as of the time of the first post.

2022-01-26_2021-10-30-fullpageos-default-bullseye-armhf-lite-0.13.0 released on 2022-01-25 18:31

The system says this is version 0.13.0

AidanHak, 2022-02-04:
To be clear, this seems to be an issue in older versions as well, which is what prompted me to re-image the Pi and update to the newest nightly build.

guysoft, 2022-02-05:
Hey,
So I just tested the 2022-02-02 nightly, and both splash screen and boot works.
I had to flash twice, not sure why. This is from the rpi imager. The first boot took a while.

I am actually thinking of putting an RC of this
[image]
[image]

MarcoBlulab, 2022-02-09:
Same problem, 
sudo apt-get update 
sudo apt-get upgrade
The Pi loads and then goes indefinitely to a black screen.
Still accessible via ssh but no out in HDMI

srcastor, 2022-04-23:
Same problem here, I have tried to restart about 20 times in a row just to make sure and on 3 times the screen stays black just after the splash image, totally random, the web that I am running is hosted in the rPI and its an empty html

**FullPageOS 0.12.0**

4o4forbidden, 2022-11-06:
Hello, @guysoft please fix this !
Or did u find a solution for this issue ? 
I'm facing the exact same issue ...

Kevinsky86, 2022-11-17:
Also running into:

> pi@fullpageos:~/scripts $ ./start_chromium_browser 
> dpkg-query: no packages found matching bluealsa
> [1117/165938.046830:ERROR:elf_dynamic_array_reader.h(61)] tag not found

Getting rid of bluealsa error seems pretty straightforward:

`sudo apt install bluealsa`

After doing this and a reboot it worked again on RPI 3

Kevinsky86, 2022-11-17:
In my case it booted to a "black screen" because chromium didn't launch. So the OS launches just fine it's the browser bit that hangs.
I don't know about it "sometimes" doing this, mine boots just fine now. I've rebooted about 5 times now to test.

4o4forbidden, 2022-11-18:
Duh, i thought u resolved it :/ !
Whatever, nice try mate ! 
Hope it will be fixed asap ... @guysoft 
as i have a project ( clients ) waiting for it to be fixed 
Else, @Kevinsky86 i used a previous version of **FullPageOS** that doesnt have this problem 
but i has another issue , that usb(s) are not working on pi 4 last version ! 
dont knw really how to explain this but there's two versions of pi4 ( 2 release dates ) with different chips 
so the ancient one ( pi4 ) works fine ( pi3 also works like a charm ) 
but not on the new pi4, would u suggest a fix plz ? 

_ 
Actually i noticed smth that all pi4 with this chip i tested usb worked fine  
[image]

but pi4 equipped with this chip usb not working
[image]

can u explain the difference please ? 

**Thanks**

jupe, 2023-11-10:
same here. doesn't work me either with "new pi4" ;
```
/scripts/start_chromium_browser 
dpkg-query: no packages found matching bluealsa
```

jupe, 2023-11-10:
> sudo apt install bluealsa

E: Package 'bluealsa' has no installation candidate

jupe, 2023-11-10:
https://forums.raspberrypi.com/viewtopic.php?t=326608

Pshock13, 2024-12-28:
I'm having the same issue. Was running fine, I did an update && upgrade, now when it powers on it loads to a white screen. I can ssh in and do `./scripts/reload_fullpageos_txt` to get the browser to load properly but that's not a real fix.

I wasn't able to locate any bluealsa installs, maybe because I'm using a RPi02W 🤷🏼

automagicalh, 2025-01-05:
same issue, black screen, booted, but no chromium