guysoft/FullPageOS #401: Current nightly does not start in kiosk mode

guysoft, 2022-02-05:
#### What were you doing?
Flash nightly 2022-02-02

#### What did you expect to happen?

Chromium should load in kiosk mode

#### What happened instead?

got a boarder of the window manager

#### Was there an error message displayed? What did it say?

#### Version of FullPageOS?

nightly 2022-0202

#### Screenshot(s) showing the problem:

[image]

Related: https://github.com/guysoft/FullPageOS/issues/400#issuecomment-1030606800

Automatica-Mat, 2022-02-12:
I have the same issue. When I change the boot page it first starts boardless and then after a minute a border appears. It's like its running the full page script again after a minute and making a border.

StanSmits, 2022-02-12:
Having same issue, The site is running fine, but the borders are here. Running version Nightly released on 2022-01-25 from raspberry pi imager

Gioz1979, 2022-02-16:
Having the same issue on 2021-04-14_2021-03-04-fullpageos-buster-armhf-lite-0.12.0 after a manual upgrade to bullseye; I tried to intercept a 'fullpagechange' event in javascript but no success

AidanHak, 2022-02-16:
Is there any way to force kiosk mode in the meantime?

StanSmits, 2022-02-16:
> Is there any way to force kiosk mode in the meantime?

"Force" just press f11 on a keyboard

dampfklon, 2022-02-24:
I think the issue is not chrome. Afaik kios mode in chrome simply hides menus like the adress bar and such.

I changed the command `matchbox-window-manager &` to ` matchbox-window-manager -use_titlebar no &` in the file `/home/pi/scripts/start_gui` which starts the windwo manager without the title bar.

guysoft, 2022-02-25:
@dampfklon thanks for the workaround. I wonder though if there is a way to make kiosk go to fullscreen. Since that setup would mean that when you press ctrl+t for a new tab, the new tab will not have any title bar neither. I haven't had the time to play with it more, since I am working on the rpi-imager fork and OctoPi first

guysoft, 2022-02-27:
Ok, so kinda amusing why this happened: chromium made ``--kiosk`` go full screen.
So the command here is not required anymore and actually exists full screen: https://github.com/guysoft/FullPageOS/blob/b1793c7344670cea3e2f6406dac95f1a352cf7d1/src/modules/fullpageos/filesystem/home/pi/scripts/run_onepageos#L6

commit 5576f3fa (2022-02-27): Fixes #401
closed 2022-02-27 by guysoft