guysoft/FullPageOS #402: Rotate Grafana Playlist

GeekyTim, 2022-02-06:
#### What were you doing?
Viewing Grafana Playlist

#### What did you expect to happen?
Playlist would rotate between views

#### What happened instead?
Stayed on first page of playlist

#### Was there an error message displayed? What did it say?
None

#### Version of FullPageOS?

0.12.0

guysoft, 2022-02-06:
Are you using fullpagedashboard? or some feature in grafana?

GeekyTim, 2022-02-06:
It's a Grafana thing, but I might have worked it out. Will know more
tomorrow.

GeekyTim, 2022-02-08:
Good news: This is not your fullpageos which is at fault, but the URLs
exposed by Grafana. I thought I had the playlist URL, but had a page URL.

Please close this 'issue'

*Thanks*

*Tim Richardson*
*Twitter: @Geeky_Tim*

guysoft, 2022-02-09:
Thanks for updating

closed 2022-02-09 by guysoft