guysoft/FullPageOS #403: Opening a URL in FullPage from a background python process

johnroberts2k, 2022-02-11:
Hi,

I've been using FullPageOS for quite some time now it's been great. I've got a new requirement where I have a python script that runs as a background process. The python script basically listens for NFC cards being touched on a device and if the NFC tag is detected it currently prints the output. However, What want to do is open a URL like for example https://myurl/?nfctag=serial but this needs to open on the FullPageOS instance that is already running. Can anyone advise how I can do this please?

My python code is as follows:

`from nfc import ContactlessFrontend
import time

def connected(tag):
    ident = ''.join('{:02x}'.format(ord(c)) for c in tag.identifier)
    print(ident)
    return False

clf = ContactlessFrontend('usb')
while True:
    clf.connect(rdwr={'on-connect': connected})
    time.sleep(1)`

guysoft, 2022-02-11:
I don't understand what aee you missing?
Setting this as a service on startup? 
That it starts before the browser does?
FullPageOS already has a webserver that loads on startup

johnroberts2k, 2022-02-11:
Yes sorry let me explain a bit better. 

FullPageOS loads a website http://myurl by default

This is fine. But i also have a python script on that same server running. And i want to be able to push a new URL from python command on the front-end FullPageOS. Does that make sense?

johnroberts2k, 2022-02-12:
Any suggestions? I need my background running script to be able to change the URL that is currently opened on fullpageos. When the Pi starts up it goes to the default set URL but then my background python script needs to be able to change the url. 

I ran terminal via SSH and when I did killall chromium-browse then fullpageos closed. What is the name of the fullpageos instance? I know in android apps have something called intent so you can open a app with the intent.

In Windows you can run chrome.exe followed by the URL. 

So just really need to know what the format to command a URL to open on fullpageos is?

Gioz1979, 2022-02-16:
You can use a websocket, connecting it from the webpage to the background script, and sending the new url from the script to the page when it needs to change

johnroberts2k, 2022-02-21:
> You can use a websocket, connecting it from the webpage to the background script, and sending the new url from the script to the page when it needs to change

That sounds very promising, any links or guides on how I can do this please? :)

thedumbterminal, 2022-03-05:
You could even use an initial page that contains a iframe or frames, then use javascript to control that inner frame, by polling (or better) a local URL to determine the URL to use in that frame.

Or run a python web server locally and point the full screen os URL to that page, then this loads a loads a frames with the required URL, then the the contain page can either be refreshed (meta refresh or JS) at intervals.

This would probably a more basic implementation to web sockets depending on your experience.

Joe2824, 2022-04-01:
run a webserver with bottle or flask. Then you can send an post request to open a url.
If you start the chrome browser with a set debug port `--remote-debugging-port=9222` you can access the running chrome with:
```
class ChromeControl(object):
    def __init__(self):
        self.service = Service(executable_path=r'/usr/bin/chromedriver')
        self.options = Options()
        self.options.add_experimental_option('debuggerAddress','localhost:9222')        
        self.check_connection()

    def check_connection(self):
        try:
            if self.driver.session_id:
                return True
        except:
            self.reconnect_browser()

    def open_url(self, url):
        if self.check_connection():
                self.driver.get(url)
```

Bluscream, 2024-07-04:
Is there no other way to change url or refresh? Can i atleast restart the browser at the same or a different url by using a combination of pkill and starting the chromium binary?

guysoft, 2024-07-04:
Yes you could kill chromium and start it.