guysoft/FullPageOS #404: no sound youtube

DesmondKH, 2022-02-17:
#### What were you doing?
i have a custom test site which shows an video from youtube

#### What did you expect to happen?
i expected the video to play with sound

#### What happened instead?
there is no sound

#### Was there an error message displayed? What did it say?
there was no error displayed 

#### Version of FullPageOS?
00.12.0
is there a way to fix this ?

guysoft, 2022-02-21:
Can you take a screesnshot? The volume sometimes is set to low by default.

Solutions I found:
1. 
Use ``raspi-config`` command to set the audio output.

2. 
According to this there is a known bug if you are using RaspberryPi 4:
https://forums.raspberrypi.com/viewtopic.php?t=274691

AFrenchCroissant, 2022-08-29:
for me, i just had to go in raspi-config (`sudo raspi-config`); then System Options (1 for now), Audio, and option 1 (for audio jack)
[Raspberry pi Documentation](https://www.raspberrypi.com/documentation/computers/configuration.html#changing-the-audio-output)

guysoft, 2022-08-29:
Thanks for the extra info. Since there was no reply to this issue I am considering closing it.

nikolaev-rd, 2024-01-14:
I want to play videos from YuoTube, but no sound through HDMI (jack didn't used).

- HDMI attached to hdmi-0 (port next after the power port) on PRi 4 (4 GB)
- `uname -a`:
   ```bash
   Linux pi-kiosk 6.1.0-rpi7-rpi-v8 #1 SMP PREEMPT Debian 1:6.1.63-1+rpt1 (2023-11-24) aarch64 GNU/Linux
   ```
- `cat /etc/os-release`:
   ```
   PRETTY_NAME="Raspbian GNU/Linux 12 (bookworm)"
   NAME="Raspbian GNU/Linux"
   VERSION_ID="12"
   VERSION="12 (bookworm)"
   VERSION_CODENAME=bookworm
   ID=raspbian
   ID_LIKE=debian
   HOME_URL="http://www.raspbian.org/"
   SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
   BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"`
   ```

What I already tried:

1. `sudo raspi-config`
   - System Options → Audio (set vc4-hdmi-0) + reboot
   - check audio device: `alsamixer`
      [image]

2. change/add options in `/boot/config.txt` in all combinations:
   - `hdmi_dirve=2` 
   - `dtparam=audio=off` (change from `dtparam=audio=on`)

3. change params in `/usr/share/alsa/alsa.conf`:
   - `defaults.ctl.card 0` to `defaults.ctl.card 1`
   - `defaults.pcm.card 0` to `defaults.pcm.card 1`

4. check `aplay -l` after step 3:
   ```
   **** List of PLAYBACK Hardware Devices ****
   card 0: Headphones [bcm2835 Headphones], device 0: bcm2835 Headphones [bcm2835 Headphones]
     Subdevices: 8/8
     Subdevice #0: subdevice #0
     Subdevice #1: subdevice #1
     Subdevice #2: subdevice #2
     Subdevice #3: subdevice #3
     Subdevice #4: subdevice #4
     Subdevice #5: subdevice #5
     Subdevice #6: subdevice #6
     Subdevice #7: subdevice #7
   card 1: vc4hdmi0 [vc4-hdmi-0], device 0: MAI PCM i2s-hifi-0 [MAI PCM i2s-hifi-0]
     Subdevices: 1/1
     Subdevice #0: subdevice #0
   card 2: vc4hdmi1 [vc4-hdmi-1], device 0: MAI PCM i2s-hifi-0 [MAI PCM i2s-hifi-0]
     Subdevices: 1/1
     Subdevice #0: subdevice #0
   ```

All that with no luck — no sound while playing videos :(

Related issue: https://github.com/RPi-Distro/pi-gen/issues/742

nikolaev-rd, 2024-01-14:
```
aplay /usr/share/sounds/alsa/Front_Center.wav

Playing WAVE '/usr/share/sounds/alsa/Front_Center.wav' : Signed 16 bit Little Endian, Rate 48000 Hz, Mono
aplay: set_params:1352: Sample format non available
Available formats:
- IEC958_SUBFRAME_LE
```

nikolaev-rd, 2024-01-14:
Sound work on RPi4 when I install: https://downloads.raspberrypi.com/raspios_lite_armhf/images/raspios_lite_armhf-2021-05-28/2021-05-07-raspios-buster-armhf-lite.zip

guysoft, 2024-06-18:
Hey, this might be a solution: 
https://github.com/guysoft/FullPageOS/discussions/553#discussioncomment-9279948

Also if you tell me it works I can add it and ship it in the next version