guysoft/FullPageOS #405: Black screen with mouse cursor

johnroberts2k, 2022-02-22:
I have fullpageos running on a Pi 3. It's working on a HDMI monitor and a TV fine. But having just plugged it into a 10.1 inch touch screen that has max resolution of 1280 x 800, the Pi Boots up but then when it gets to the OS I get a black screen with a mouse cursor. I can move the mouse cursor around but the Browser doesn't launch. 

At first it was only showing splash screen of the Pi and then video signal was getting lost. I then had to change config.txt to enable hdmi_safe, after I enabled hdmi_safe then the Pi boots up fine but the FullPageOS doesn't launch.

johnroberts2k, 2022-02-22:
I installed this version "fullpageos-buster-armhf-lite-0.12.0" and now it loads on the 10.1 inch touch screen. but when I am using my touh screen it would just completely cut-out the display output randomly in the middle and the monitor shows no signal message. I can still connect via SSH and reboot it. When I reboot it then it works until i start using the display again and same thing repeats. What could be wrong?

chvvkumar, 2022-02-24:
I am having the same issue with a Pimoroni HyperPixel 700x700 square screen with a Pi:

```
pi@fullpageos:~ $ cat /etc/debian_version
10.11
pi@fullpageos:~ $ cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

pi@fullpageos:~ $ uname -a
Linux fullpageos 5.10.63-v7+ #1496 SMP Wed Dec 1 15:58:11 GMT 2021 armv7l GNU/Linux

pi@fullpageos:~ $ cat /proc/cpuinfo
processor	: 0
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 89.60
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 1
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 89.60
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 2
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 89.60
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 3
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 89.60
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

Hardware	: BCM2835
Revision	: 9020e0
Serial		: 000000005b61f273
Model		: Raspberry Pi 3 Model A Plus Rev 1.0
```

[image]

guysoft, 2022-02-24:
Looks like its not starting due to no internet connection.
The run script waits for internet before it starts the browser, so you will not get a disconnected page error message.
This is the script that does that:
https://github.com/guysoft/FullPageOS/blob/devel/src/modules/fullpageos/filesystem/home/pi/scripts/run_onepageos#L5

chvvkumar, 2022-02-24:
> Looks like its not starting due to no internet connection. The run script waits for internet before it starts the browser, so you will not get a disconnected page error message. This is the script [...]

The Pi did connect to WiFi before the UI came up completely. I was able to SSH to the Pi before the UI came up. For reference, I am using a local webpage hosted on the same LAN on another docker host and not an internet page.

I will try disabling the scrit and see if it helps.

aberkvam, 2022-04-09:
That script seems to be checking the HTTP response code and only accepting a "200".  Anything else is regarded as a failure and the screen stays black.

I'd propose that a response of any sort should be regarded as a successful test and the page should be displayed.  At the very least it would allow the user to see an error.

Or any 2xx or 3xx response should be accepted (which would cover redirections, for example).  And 401 should be accepted so a user can be presented with an opportunity to enter a user and password.

guysoft, 2022-05-09:
@aberkvam You are welcome to send a pr. The script and exact line is located here:

https://github.com/guysoft/FullPageOS/blob/devel/src/modules/fullpageos/filesystem/home/pi/scripts/run_onepageos#L5

YAMLcase, 2022-06-01:
> @aberkvam You are welcome to send a pr. The script and exact line is located here:
> 
> https://github.com/guysoft/FullPageOS/blob/devel/src/modules/fullpageos/filesystem/home/pi/scripts/run_onepageos#L5

Since I ran into the same issue (and fixed it before finding this issue), here's a PR. @aberkvam see if this is what you had in mind.

#430

guysoft, 2022-06-01:
Merged and started a nightly build, thanks!
Should I leave this open for @johnroberts2k  or should I just close this?

uprnick, 2023-09-05:
Hey all, 

Having the same issue above, even after the merge from the PR. 

- Running the 2023-05-12 (Nightly) release - from the Raspberry Pi Imager
- Sudo nano'ed into `/boot/fullpageos.txt` and changed the URL to just be one URL
- URL is on our intranet site (an Angular 13 page), not sure if that's throwing off the error run above 

Getting a black screen with mouse arrow centered to the screen.

guysoft, 2023-09-05:
@uprnick Can you connect to the device and see if the internet is working? FullPageOS waits for an internet connection to the page before opening the browser.

uprnick, 2023-09-05:
> @uprnick Can you connect to the device and see if the internet is working? FullPageOS waits for an internet connection to the page before opening the browser.

Hey, thanks for the quick response. Yes the device is connected, and verified. It is on a separate VLAN from my pc but was able to validate. 

Is there an easy way (like above) to disable that new code/check?

guysoft, 2023-09-07:
Its looking at what is stored at ``/boot/fullpageos.txt``. the code is here:
https://github.com/guysoft/FullPageOS/blob/devel/src/modules/fullpageos/filesystem/home/pi/scripts/run_onepageos#L5

Which calls:
https://github.com/guysoft/FullPageOS/blob/devel/src/modules/fullpageos/filesystem/home/pi/scripts/get_url

bluecraank, 2023-11-06:
Just noticed it also is not working if webpage has ssl error
So i mounted the image, added "--ignore-certificate-errors" in chromium startup and changed text to "disabled" in /boot/check_for_httpd

Maybe there should be a option to disabled certificate check (curl --insecure) and for chromium "--ignore-certificate-errors"

guysoft, 2023-11-06:
We could add it to the FAQ. But this is a security vulnerability and therefore should not be the default behavior.
You can also try and run an update to the certificates authorities:
```sh
apt-get install -y ca-certificates
update-ca-certificates
```

bluecraank, 2023-11-06:
> We could add it to the FAQ. But this is a security vulnerability and therefore should not be the default behavior. You can also try and run an update to the certificates authorities:
> 
> ```shell
> apt-get install -y ca-certificates
> update-ca-certificates
> ```

Of course this should only be done if site is trusted. We have our own CA for certificates, so by default its not trustet.
So update-ca-certificates wont work until i added our root-ca

guysoft, 2023-11-06:
Thats cool.
You are welcome to add that to the wiki if you think there is a good flow to share.

If you are using this with employees and FullPageOS helping your life, your organization or makes you happy, please consider making a donation. Since I am maintain this 100% as a volunteer.

Baggerfahrer, 2024-04-26:
> Just noticed its not working if webpage has ssl error So i mounted the image, added "--ignore-certificate-errors" in chromium startup and changed text to "disabled" in /boot/check_for_httpd
> 
> Maybe there should be a option to disabled certificate check (curl --insecure) and for chromium "--ignore-certificate-errors"

many thanks for this help, helped me too to display a self-signed ssl https site.
thanks!

guysoft, 2024-06-05:
Latest RC1 for 0.14.0 should fix this issue.

closed 2024-06-05 by guysoft