guysoft/FullPageOS #409: Cannot disable automatic multitab switching feature

dudfunk, 2022-03-01:
#### What were you doing?
I have multiple tabs on the fullpageos dashboard and I don't want them to automatically switch as it does by default.  The Pi is connected to a touch screen and the clients will switch tabs as needed.  I have to click the pause button if the device is rebooted.  There should be a way to disable this but I don't know how or where to look for a config file.  Also, after I pause the auto tab switch, if I add another tab, it "un-pauses" itself and I have to pause again.

#### What did you expect to happen?
I expected a way to disable the auto tab switch slideshow-like function.

#### What happened instead?
Tabs will switch unless I press the pause button.

#### Was there an error message displayed? What did it say?
No errors.

#### Version of FullPageOS?
fullpageos_version 0.12.0
Raspberry Pi 4 Model B Rev 1.2

guysoft, 2022-03-01:
Hey,
The tab management is done with [FullPageDashboard](https://github.com/amitdar/FullPageDashboard/). Its not maintained anymore, so you can either add that option as a PR or manually edit the code, or drop usage of it altogether, but then you won't be able to have multiple tabs.
What is the user case where you want tabs but you don't them to switch automatically? I mean it means there is some user interaction going on with the webpage.

dudfunk, 2022-03-01:
For now only one page is being requested to be displayed, but in case they want an additional tab, I'd rather not have the tabs cycle.  Instead, have the user switch tabs manually.  It's all good once I hit pause, but say there's a power outage, and the pi reboots.  I guess I can just tell the user to hit pause when necessary.  Where would I find the code to disable the auto switch function?

guysoft, 2022-03-01:
Who is they?
You could just update the tab.

dudfunk, 2022-03-01:
"They" refers to the department using the Pi for the webpage displayed on a TV.  
Update the tab?

guysoft, 2022-03-02:
@dudfunk Uncommenting this line can help:
https://github.com/amitdar/FullPageDashboard/blob/master/index.php#L159

You can also change the default rotate time in the config.
Its located at ``/var/www/html``

I am taking this time to remind you that FullPageOS is maintained only by me as 100% volunteer work. I don't get paid by anyone.
If FullPageOS is helping you in your workplace please consider making a donation from the link in the main page of this repo.

dudfunk, 2022-03-02:
I understand and appreciate your work.  I work in the public sector and my agency and the ones I service do not make a profit.  I don't expect support from authors of free software.  I didn't know where to just ask about the auto switch, it's not really an issue.  I saw on your page your social media contacts, I'm not on social media.  Thanks again, I'm closing this thread.

closed 2022-03-02 by dudfunk

guysoft, 2022-03-02:
Asking here is the right place :)