guysoft/FullPageOS #410: Ability to not disable DPMS and screen blanking

thedumbterminal, 2022-03-05:
#### What were you doing?

Running a raspberry pi with a touchscreen, but I want the display to turn off when idle as normal.

#### What did you expect to happen?

A configuration option to stop DPMS and screen blanking.

I know this distro is for kiosk operation so maybe disabling DPMS would be the default and any such option would stop this behaviour.

#### What happened instead?

I had to remove the first few lines of the script:

/home/pi/scripts/start_gui

However I can't find the source of this file in: https://github.com/guysoft/FullPageOS/tree/devel/src/modules/fullpageos/filesystem/home/pi/scripts

#### Version of FullPageOS?

0.12.0

guysoft, 2022-03-05:
You are welcome to send a Pull Request with a script that disables it. I am not implementing it because I have no time for it.

thedumbterminal, 2022-03-05:
No problem, do you know where the source of `/home/pi/scripts/start_gui` resides in github?

guysoft, 2022-03-05:
> No problem, do you know where the source of `/home/pi/scripts/start_gui` resides in github?

Here: https://github.com/guysoft/CustomPiOS/blob/devel/src/modules/gui/filesystem/home/pi/scripts/start_gui

You can place a script at ``modules/gui/filesystem/home/pi/scripts``. 
That lets you switch if off, or on/off.

thedumbterminal, 2022-03-06:
I've put a PR into CustomPiOS as this is not a specific FullScreenOS issue. Let me know if the approach is suitable, thanks for the quick replies.

guysoft, 2022-03-06:
Merged the changes, also thanks for the rotation option touch.
Should this issue be closed?

thedumbterminal, 2022-03-06:
yup, nothing left to do here, thanks.

closed 2022-03-06 by thedumbterminal