guysoft/FullPageOS #412: FullPageOS 0.13.0 RC1 status

guysoft, 2022-03-07:
FullPageOS 0.13.0 first release candidate.

Features:
* Chromium Version 98.0.4758.106
* RpiOS 2022-01-28
* Support to for hardware raspberrypi that have been popping up 
* Fixed  Chromium Feature Flag "TranslateUI" was renamed to "Translate" #375  (Thanks @trippled1993)
*  Add emoji support, fixes #378 (Thanks @SimonIT and @havenonick for reporting)
*  Improve chrome refesh after a crash #367 (Thanks @yrien30)
* New ``reload_fullpageos_txt`` script added when updating fullpageos txt file #391 (Thanks @shaunhurley)
* Allow power management to be disabled and screen rotated https://github.com/guysoft/CustomPiOS/pull/157 (Thanks @thedumbterminal)

## Download
https://unofficialpi.org/Distros/FullPageOS/nightly/2022-03-07_2022-01-28-fullpageos-bullseye-armhf-lite-0.13.0.zip
md5: [``18eb778724db18af7fe934adb7a79832``](https://unofficialpi.org/Distros/FullPageOS/nightly/2022-03-07_2022-01-28-fullpageos-bullseye-armhf-lite-0.13.0.zip.md5)

## no-acceleration variant

http://unofficialpi.org/Distros/FullPageOS/no-acceleration/nightly/2022-03-07_2022-01-28-fullpageos-no-acceleration-bullseye-armhf-lite-0.13.0.zip
md5: [``ded199d21e28336386a14aa19793dfd4``](https://unofficialpi.org/Distros/FullPageOS/no-acceleration/nightly/2022-03-07_2022-01-28-fullpageos-no-acceleration-bullseye-armhf-lite-0.13.0.zip.md5)

TiWe1998, 2022-03-14:
Unfortunately I have a problem with the build. After exactly 10 minutes, the screen goes black. According to the Internet probably a screensaver. However, I should first install one to deactivate it... Can't you solve it differently?

thedumbterminal, 2022-03-14:
This is the script that disables the power management: https://github.com/guysoft/CustomPiOS/blob/devel/src/modules/gui/filesystem/home/pi/scripts/start_gui

If you run that script manually on your device it should execute without error and disable power management, jut like it should on boot up.

Kevinsky86, 2022-11-17:
Is this still beeing worked on?

guysoft, 2023-01-26:
Hey, yes it is, I am just swamped with work and did not get time to get to it at all.
If it works for you It would help me determine if its stable. At the moment I only had people saying it does not work.

kevin2xk, 2023-06-12:
Hi, does anyone know how to fix the standby after 10 minutes? I have the same issue.
Will the above script fix the issue permanently?

thedumbterminal, 2023-06-12:
@kevin2xk are you running a window manager? If so then the start_gui script should be being run on start up, this will disable screen blanking

thedumbterminal, 2023-06-12:
does your matchbox-window-manager start correctly? (trying to check if this script is being used)

kevin2xk, 2023-06-12:
I'm sorry, I don't know how I can check this. I'm new to Raspberry Pi and Unix. :(
I tried to run the three xset commands from the start_gui file manually and this is the output: xset:  unable to open display ""

thedumbterminal, 2023-06-12:
Does your raspberry pi have a screen? if so does it show windows and icons etc?

You cant run the above commands if you are ssh'ed into your pi as it will need to be from a console window via your raspberry screen

guysoft, 2023-08-20:
I think at this point it can be released.

guysoft, 2023-10-22:
Released
Sorry for the delay:
https://github.com/guysoft/FullPageOS/releases/tag/0.13.0

closed 2023-10-22 by guysoft