guysoft/FullPageOS #414: Some small problem with latest version

TiWe1998, 2022-03-18:
Hello.

I have now tested the version from March 17.03.2022 and I noticed some small things.

First of all: Saving the "pin" in the bar at the top works. That's great!

I also noticed this:

1. FullPageOS is not started. After the standard splash screen, only a black screen was visible. After running "sudo apt update/upgrade" it worked.
2. There is still a small white bar at the bottom of the website. This runs through all pages.
3. The splash screen does not appear at all. Only black until the "right" start.

Thank you for your awesome work!

guysoft, 2022-03-20:
1. Do you know what the upgrade changed? Can you test again?
2. can you post a screenshot?
3. Might be something was changed in bullseye. I will have to look in to it, but not sure when I will have time.

jsponz, 2022-11-07:
@TiWe1998, I am in the same situation as you. Have you managed to fix it? 

Thanks in advance.

jsponz, 2022-11-08:
Thanks for your reply. 

I installed the last nightly build and it works as you describe.

4o4forbidden, 2022-11-08:
i also installed the last nightly build but sometimes it boots a black screen ! 
did u managed to fix that plw ? @jsponz  @TiWe1998

jsponz, 2022-11-08:
Actually, I installed it yesterday and tested it a couple of times. I will test it during the weekend. 

In my case, it takes up to one minute, but it loads...

4o4forbidden, 2022-12-09:
@jsponz  any news plz ?

jsponz, 2022-12-10:
Sorry, very busy at work. 
I plug it again after some weeks. First time I got the blank screen :-( 
I ran apt update & upgrade. Rebooted. It works again fine. I have powered it off for one hour. Boot it again and it works. Keep testing though.

4o4forbidden, 2022-12-11:
@jsponz thanks bro for your feedback ! 
can u do some more extra tests for me plz ? 
like more 10 times if u can ?
so i can be more sure that updating and upgrading the os fix the issue !

Thanks a lot !

jsponz, 2022-12-11:
I have tested 10 times after being off for 10 hours. Basically, it runs for 10 minutes and then I reboot it. Everything ok. 

I have switched it off for 10 minutes and started again. Everything ok. 

I hope that helps!

4o4forbidden, 2022-12-11:
Thanks a lot bro ! 
U re awesome

guysoft, 2023-04-05:
Can I close the issue?