guysoft/FullPageOS #415: Youtube video doesn't play

Carlos-MVP, 2022-03-23:
#### What were you doing?

I have a wordprees site which shows a video from youtube and a few images

What did you expect to happen?
I expected the youtube video to play automatically.

What happened instead?
The video does not start, until I click.

Was there an error message displayed? What did it say?
there was no error displayed

Version of FullPageOS?
00.12.0
is there a way to config this ?

guysoft, 2022-03-23:
Can you provide a url? I can't test without an example

Carlos-MVP, 2022-03-25:
I already solved my problem, it's nothing related to fullpageos.
I just had to enable sound permission on my website from chromium settings.

Thanks for your attention, the project is great.

closed 2022-03-25 by Carlos-MVP