guysoft/FullPageOS #416: Screensaver fullpageos

RogerSero, 2022-03-27:
Hello, how can I activate the screensaver in fullpageos?

guysoft, 2022-03-27:
You mean blanking the screen? or anything else?

thedumbterminal, 2022-03-27:
You can enable the default power management features of the OS by setting the following environment variable in `/home/pi/scripts/start_gui`:

```
DISABLE_POWER_MANAGEMENT=no
```

guysoft, 2022-03-28:
``DISABLE_POWER_MANAGEMENT=no`` works only in the nightly builds. Thanks for posting answers.  Once its in stable adding it to the wiki would be a good idea

RogerSero, 2022-03-28:
the truth, I was thinking of turning off the screen after a while, and turning it on by touching it, would this be possible?

guysoft, 2022-03-28:
Solved, see above.

closed 2022-03-28 by guysoft

RogerSero, 2022-03-29:
When I try to edit-it, it gives me this error message:

(chromium-browser-v7:6162): Gtk-WARNING **: 10:20:10.048: cannot open display:
Error: Can't open display: (null)
Failed creating new xdo instance
dpkg-query: no packages found matching bluealsa

RogerSero, 2022-03-29:
I want to know where I have to edit to activate the screen saver?  if I enter this command home/pi/scripts/start_gui, I get the above mentioned error

guysoft, 2022-03-29:
Use the nightly build and follow the instructions already posted to you: https://github.com/guysoft/FullPageOS/issues/416#issuecomment-1080035635