guysoft/FullPageOS #418: Reboot and shutdown via ssh

RogerSero, 2022-04-04:
Hi, I use fullpageos to display my home assistant panel. I have created a button to be able to restart and shutdown fullpageos through ssh. When I restart, the screen does not start, it stays black, instead through putty yes. That could be happening?

guysoft, 2022-04-04:
I don't know, can you please explain how did you implemented the button?

closed 2022-04-06 by RogerSero

guysoft, 2022-04-06:
Hey, would you like to share how you solved the issue?

RogerSero, 2022-04-07:
The problem is that I'm using an rpi3 A+, and reading I saw that for this board the best thing was to use the no-acceleration version. So all my problems were solved. I was able to use node-red to issue ssh commands to shutdown and reboot the board, without crashing.

guysoft, 2022-04-07:
Where did you read that?
I don't remember documenting that and would be interested to see if that helped anyone.
Since I was considering phasing it out

RogerSero, 2022-04-07:
Reading the release notes, in 0.10, you added support for the raspberry pi A+ and noted that you created the no-acceleration version for those boards that needed to turn off the gpu to not consume so many resources.  That's when I did the test with that version and it worked for me.

Bart1909, 2023-11-22:
Hi,
I've got the same problem, running on a Pi 4.
When I power off the pi and power on, the browser starts fine.
But when I ssh into the pi and run "sudo reboot" it reboots but the display stays black.

I'm using the latest release with version 0.13.0

Any ideas?

guysoft, 2024-01-15:
@Bart1909 That does not seem to be the same issue described here. Because it talks about using a Pi A+ on the non-acceleration variant.

Your issue sounds like a display issue - did you try a different display?