guysoft/FullPageOS #421: Screen not rotating

BetelCode, 2022-04-28:
#### What were you doing?

I am trying to show a single rotated local webpage using the "display_hdmi_rotate" command. I also tired to disable "dtoverlay=vc4-kms-v3d" it worked, But the webpage turned white and I couldn't see the webpage.  I also looked at older issues about the rotation issue but those fixes did not work.

#### What did you expect to happen?

I expected the page to be displayed the right way and rotated.

#### What happened instead?

The screen will not rotate. / The webpage turned white.

#### Was there an error message displayed? What did it say?

None.

#### Version of FullPageOS?

same issue on Fullpage OS stable & and installed on the Latest Nightly build, But the issue still persists.

guysoft, 2022-05-01:
What display are you using?
There is an issue with the official display, see https://github.com/guysoft/FullPageOS/issues/247#issuecomment-443470254

thedumbterminal, 2022-05-05:
Have you tried changing this variable: https://github.com/guysoft/CustomPiOS/blob/devel/src/modules/gui/filesystem/home/pi/scripts/start_gui#L7

guysoft, 2022-05-12:
You go in the filesystem of the pi to the path ``/home/pi/scripts/start_gui`` explaining how to do that is beyond the scope of FullPageOS.

H3npi, 2022-05-25:
@BetelCode in /boot/config.txt I set `display_hdmi_rotate=3` for a 270 degree turn on the hdmi output

thfrei, 2022-05-30:
Ok, I spent nearly an hour trying it. As it turns out it makes a difference if you use a RPI4 or older modell.

Explanation: It seems to me, that if you are using a RPI4 (raspberry pi 4) you must comment all `dtoverlay` and make sure, that you do NOT use the vc4-kms-v3d or vc4-fkms-v3d. These vc4-x are gpu accelarated drivers using opengl. Opengl does NOT respect the rotate option in config.txt, since it has its own config tool using the desktop gui of raspian.

Solution:
- For RPI4 make sure to use legacy driver mode. This can be done by not having any active configuration for `dtoverlay`
- use `display_hdmi_rotate=1`
- you should increase gpu_memory split, since default split is 16mb and this is not sufficient for a vertical screen.

My complete config.txt:

```config
# For more options and information see
# http://rpf.io/configtxt
# Some settings may impact device functionality. See link above for details

# uncomment if you get no picture on HDMI for a default "safe" mode
#hdmi_safe=1

# uncomment this if your display has a black border of unused pixels visible
# and your display can output without overscan
#disable_overscan=1

# uncomment the following to adjust overscan. Use positive numbers if console
# goes off screen, and negative if there is too much border
#overscan_left=16
#overscan_right=16
#overscan_top=16
#overscan_bottom=16

# uncomment to force a console size. By default it will be display's size minus
# overscan.
#framebuffer_width=1280
#framebuffer_height=720

# uncomment if hdmi display is not detected and composite is being output
#hdmi_force_hotplug=1

# uncomment to force a specific HDMI mode (this will force VGA)
#hdmi_group=1
#hdmi_mode=1

# uncomment to force a HDMI mode rather than DVI. This can make audio work in
# DMT (computer monitor) modes
#hdmi_drive=2

# uncomment to increase signal to HDMI, if you have interference, blanking, or
# no display
#config_hdmi_boost=4

# uncomment for composite PAL
#sdtv_mode=2

#uncomment to overclock the arm. 700 MHz is the default.
#arm_freq=800

# Uncomment some or all of these to enable the optional hardware interfaces
#dtparam=i2c_arm=on
#dtparam=i2s=on
#dtparam=spi=on

# Uncomment this to enable infrared communication.
#dtoverlay=gpio-ir,gpio_pin=17
#dtoverlay=gpio-ir-tx,gpio_pin=18

# Additional overlays and parameters are documented /boot/overlays/README

# Enable audio (loads snd_bcm2835)
dtparam=audio=on

[pi4]
# Enable DRM VC4 V3D driver on top of the dispmanx display stack
# disabling, since fkms does not allow rotation via config.txt
# https://www.raspberrypi.com/documentation/computers/config_txt.html#display_hdmi_rotate
#dtoverlay=vc4-fkms-v3d
max_framebuffers=2

[all]
# enable raspicam
start_x=1
# gpu_mem=128
disable_splash=1
# disable all new rpi4 display driver, use legacy by commenting out dtoverlay
# https://www.raspberrypi.com/documentation/accessories/display.html
#dtoverlay=vc4-kms-v3d

# vertical screen
# increase gpu_mem for display rotate
gpu_mem=256
# Rotate screen to horizontal
# 1 is 90 degrees. 2 is 180 degress. 3 is 270 degrees.
display_hdmi_rotate=1
# only for lcd screens
# display_lcd_rotate=1
# only used for syncing touch wich screen
# lcd_rotate=1
```

Sources: 
- https://www.raspberrypi.com/documentation/accessories/display.html
- https://www.raspberrypi.com/documentation/computers/config_txt.html#display_hdmi_rotate

thfrei, 2022-05-30:
I updated the FAQ: https://github.com/guysoft/FullPageOS/wiki/FAQ#how-to-rotate-the-screen-to-a-vertical-position

BetelCode, 2022-06-16:
> Have you tried changing this variable: https://github.com/guysoft/CustomPiOS/blob/devel/src/modules/gui/filesystem/home/pi/scripts/start_gui#L7

This was the fix thank you!

closed 2022-06-16 by BetelCode

Axelf93, 2022-08-30:
For anyone that might find this useful I tried the FAQ on a Rasberry-Pi 3B after a lot of attempts and failiures I managed to get it working. Install FullPageOS on RPI and clean install - Let it boot and restart. 
Go
nano  /boot/config.txt

and add # infront of any you might see:  dtoverlay=vc4-kms-v3d & dtoverlay=vc4-fkms-v3d

Add display_hdmi_rotate=3 or any number you want and Viola

guysoft, 2022-10-28:
Thanks for adding to the wiki! :)
https://github.com/guysoft/FullPageOS/wiki/FAQ#how-to-rotate-the-screen-to-a-vertical-position