guysoft/FullPageOS #425: Issues with "Signage" mode

abaddononion, 2022-05-09:
This may not be a full feature yet because it's commented out in the start_chromium_browser script, but I figured I'd make some mention of it anyway, at least as documentation for other users. 

When attempting to enable Signage mode, so that FullPageOS can restart chromium on ERROR/FATAL conditions, the tail command is started too fast (at least on my hardware), and winds up following an empty/missing file. To fix this, simply inserting a "sleep 30" after launching the backgrounded Chromium and before entering the while loop works fine. However it's worth noting that the length of this sleep command is also the length of time errors could get missed during Chromium startup... depending on your ideology for how Signage mode should work, the sleep might should be reduced. You could also remove the -n 0 from the tail command, but this will cause issues in the case that run_onepages regenerates a new instance of start_chromium_browser. To fix this I would probably add a line within run_fullpages to nuke /home/pi/.config/chromium/chrome_debug.log before firing off start_chromium_browser. But these are all just loose suggestions. 

Additionally, because of the logic of this block:

`tail -n 0 -f "$logfile" | while read LOGLINE &> /dev/null; do

   echo "Refreshing after crash"
   echo "Restarting at `date` after a reported crash. Logline: ${LOGLINE}" >> /tmp/crashlog
   [[ ("${LOGLINE}" == *"ERROR"*) || ("${LOGLINE}" == *"FATAL"*) ]] && /home/pi/scripts/refresh

done`

The "restarting" messages are printed for EVERY line written to the debug log, regardless of whether or not it contains ERROR or FATAL, meaning it always writes a message to crashlog claiming it's performing a restart even when it's not. This only comes up with certain error messages that output additional lines... but it's confusing behavior when one runs into it. Ideally the test conditions there would wrap all three lines of "action". 

I'm further running into an additional issue which I haven't been able to solve yet, and may be unique to the site I'm trying to load... but I'm getting an error like this:

`[722:751:0509/115119.942467:ERROR:ssl_client_socket_impl.cc(962)] handshake failed; returned -1, SSL error code 1, net_error -101`

Which is merely an SSL error. This happens every time the page is loaded, so my system is crash-looping. I've tried adding the following flags:

`--ignore-certificate-errors --ignore-ssl-errors --ignore-certificate-errors-spki-list`

So far that hasn't worked out for me. I'll keep digging into the nuances of my situation, but the main issue I'd like to raise here is that clearly catching all "ERROR" messages is a tad overzealous. Either certain types of error need to be ignored, or the condition above needs to be restricted to only look for "FATAL" messages. I don't know enough about chromium's behavior at this point to make an educated call, and I'm still digging on this one myself.

abaddononion, 2022-05-10:
> What url are you using which makes you get an ssl error?

It's an internal developer site provided by another team, so unfortunately
I have limited insight into what exactly it's doing.

I wound up solving this problem for us by piping the tail to `grep -v
'ERROR:ssl'`. A very wonky answer, but it allowed us to move on.

abaddononion, 2022-06-10:
@MalteP That's great information, I really appreciate the explanation on why that chromium flag wasn't working for me, and hadn't realized it's because of a self-signed cert. I'm going to dig into that more in our environment.

Unfortunately your specific method won't help in our scenario, because we're using automation to deploy about a hundred Pis worldwide with various URLs, and I'm trying not to do any customizations that requires specific tweaks for each thing. So I'll stick to the hack I came up with previously (grepping out ssl errors specifically). But maybe the information you've provided here will help someone think up a more all-encompassing solution.

MalteP, 2022-06-10:
If you use one Root CA to generate all your internal certificates, you may just add it to the systems certificate store (/etc/ssl/certs) which is used by curl and chromium. The Debian Linux way to do this is to copy the public certificate to /usr/local/share/ca-certificates/ and use sudo update-ca-certificates to import it. No other "hacks" needed, but i did not check it yet.

If they use a different CA and you just need to bypass the check, maybe implement the generation of the fingerprint in the existing bash script. I guess the openssl s_client can be found in the image. This could be a interesting feature for some use cases, i'll try to implement it within the next week if i have some spare time.

Thanks for the information on the check_for_httpd. I was focused on the SSL issue and did not realize that this check can be disabled. :-)

MalteP, 2022-06-22:
I was experimenting with the --ignore-certificate-errors-spki-list option and the automatic generation of the Base64 stuff. 

One of our internal systems did not work with this, but needed the --ignore-certificate-errors flag instead, which i thought would have no effect. As i had no time to take a closer look yet, here's my modified start_chromium_browser script, which hopefully will be useful to someone.

The script checks for /boot/ssl_trust_selfsigned which enables the hack. Also either the -k curl option or a "disabled" in /boot/check_for_httpd is needed.

[start_chromium_browser.txt](https://github.com/guysoft/FullPageOS/files/8960572/start_chromium_browser.txt)

guysoft, 2022-06-28:
Ok, thanks! closing

closed 2022-06-28 by guysoft