guysoft/FullPageOS #427: PC Hardware?

mikopia, 2022-05-16:
Can we install FullPageOS on PC hardware (i.e. Dell OptiPlex Micro)? Given that it is hard to find RPis these days.

Or what other alternatives are there?

guysoft, 2022-05-16:
You could build an armbian variant (which is supported in CustomPiOS) and run it on all the devices supported here: https://www.armbian.com/
If there is some kind of debain x64 base image you could use that.

However I haven't tested that in a while so if you hit any issue please report it.

Who is we?

kevireilly, 2022-09-12:
I’m also interested in something like FullPageOS for a non-raspberry pi device. @mikopia were you able to get something working or identify an alternate solution?

guysoft, 2022-09-12:
Related or duplicate: https://github.com/guysoft/FullPageOS/issues/436

guysoft, 2022-09-12:
@kevireilly non-raspberrypi or PC? BEcause there is a difference. Armbian could be built and adapted, but running on PC x86 would require major changes to the build system

kevireilly, 2022-09-12:
Fair question and essentially both, though x86 seems potentially more common at a lower price point and smaller form factor.

guysoft, 2022-09-12:
That depends, Single Board computers are commonly arm64 and cheaper,. also you save on power. But if you are looking in to re-using old laptops then they might be more available around you.

Although it would be great to support both. ATM the build system has limitations to build for stuff which is not a Debian-based image that you can chroot in to. It would need someone to step up and add support, since I have no time for this ATM.

kevireilly, 2022-09-12:
Ah true true that this is a different scenario for SOCs and such. While @mikopia’s underlying issue was mentioned as hardware availability, in addition to that I’m also looking at compute/graphics power relative to cost for an image and video use case. Though power requirements are an important portability point too for sure.

At any rate, really appreciate you providing feedback, contributing really useful open source software, and if I make any progress on broader support I will definitely be contributing. Thanks very much.

kevireilly, 2022-09-24:
This may be one option: https://porteus-kiosk.org/

guysoft, 2022-10-18:
@kevireilly They support intelx86 only and not Raspberrypi and other arm devices. The fact they mention in likely means they have an open issue on their systems to support Raspberrypi/ARM. I guess its a mirror of this issue :)