guysoft/FullPageOS #428: Auto Login to a Page

TjOsT1975, 2022-05-24:
i´m using 0.11 on a pi 3 and try to load a internal website report.
while using fullpageos.txt the site loads fine. but somehow it did not load anymore on fullpagedashboard.txt.

first it did not load the sites in fullpagedashobard.txt and no auto login via web address is possible anymore. I do not know what happend but i do have 5 pi´s with fullpagedashboard that are working fine. 
The site are blank.
with single site there comes a user and password dialog and afterwards it is working.

We do have also zscaler might that be a problem?
site looks like this: http://user:password@jasperreports...
i can not figure out why it is not working anymore.

guysoft, 2022-05-25:
Could it be an https certificate thing? Can you post output of the network console of something with more info than "it does not work"?

TjOsT1975, 2022-05-25:
can you tell me what you need and how i can do it?

guysoft, 2022-05-25:
1. the url
2. the output ftom network and console when using FullPageDashboard https://developer.chrome.com/docs/devtools/network/

TjOsT1975, 2022-05-25:
i can not give you the url. it is in our intranet.
i can not open dev tools. 
Sorry.

TjOsT1975, 2022-05-27:
i can confirm that in FullPageDashboard.txt website with username and password like:
http://user:password@jasper.html
do not load or show any login window.
with the removing of username and password the site will load.
my problem is that the automatic function is now not working and will make problems
is there a way to fix this issue?

guysoft, 2022-10-18:
That format is basic http auth, and doing it in the URL has been deprecated on chromium's side:
https://medium.com/@lmakarov/say-goodbye-to-urls-with-embedded-credentials-b051f6c7b6a3

FullPageOS does not aim to recompile chromium atm to enable that might be useful to the project but deprecated.

closed 2022-10-18 by guysoft