guysoft/FullPageOS #432: Chanigng the user in rpi-imager from breaks system

guysoft, 2022-06-20:
Continuing report of @AlbertMN here: 
https://github.com/guysoft/FullPageOS/issues/276#issuecomment-1159389974
It seems that the userless new setup breaks fullpageos if the username is changed.

Related:
https://github.com/guysoft/CustomPiOS/issues/163

jory3, 2023-12-12:
In case someone else needs a quick fix for the "unable to launch "/home/pi/scripts/start_gui" X session"-error: I had created a user 'myuser' in the pi imager. After the first boot I added a user called 'pi' manually and copied the folder /home/myuser/scripts to /home/pi/scripts. Then I did a `chmod pi:pi /home/pi -R`and rebooted, now everything works.

rameshnavi, 2024-02-27:
Faced same issue. 

> In case someone else needs a quick fix for the "unable to launch "/home/pi/scripts/start_gui" X session"-error: I had created a user 'myuser' in the pi imager. After the first boot I added a user [...]

This worked(kind of) for me. Had to do `chown` instead of `chmod`

guysoft, 2024-02-28:
Related #525

Also started having a go at fixing this. Needs changes both in CustomPiOS and FullPageOS. I think I got the CustomPiOS side for now

w1s3on3, 2024-03-06:
i fixed mine just by creating a link, 
`ln -s /home/monitor /home/pi`

Pshock13, 2024-12-15:
I did a fresh install today from raspberry pi imager. The full quick fix I did was to just make the directory for pi and then symlink to the actual file in my user's home directory. change some permissions, reboot.

```
mkdir -p /home/pi/scripts
ln -s /home/USER/scripts/* /home/pi/scripts/
chown -R USER:USER /home/pi/scripts
```

Obviously an easier work around would be to just use `pi` as the user when going through the setup. But if you're like me, you have a username that you use for all your other devices and don't want to be adding more :)