guysoft/FullPageOS #433: Autofill / suggestions help - new to Pi

baillyj, 2022-07-21:
#### What were you doing?
Running page in fullpageOS, browser is showing suggestions for previously entered users in webpage text field 

#### What did you expect to happen?
Using this for our online timeclock and want to make sure nobody's username populates when clicking on the text field. 
#### What happened instead?
As of right now, the browser saves suggestions after a user logs in. Clicking hide suggestions only hides the window that appears and will show up once a user clicks on the text field again. 
#### Was there an error message displayed? What did it say?
No error as it is running as expected. Just need some assistance in possibly customizing the behavior to fit our needs and not sure how to go about this as I am a beginner to Pi building. 
#### Version of FullPageOS?
Just built this SD card so hopefully most current - shows 0.12.0 in fullpageos_version

#### Screenshot(s) showing the problem:
#
[image]

[If applicable. Always include if unsure or reporting UI issues.]

#### If you are building FullPageOS - provide a build.log that is created for the build

guysoft, 2022-07-24:
You can disable autocomplete by doing this:
1. In Chrome > Settings > Advanced > Passwords and Forms, uncheck "Enable Autofill to fill out web forms in a single click"

Source:
https://bugs.chromium.org/p/chromium/issues/detail?id=468153

guysoft, 2022-07-24:
Related: https://github.com/guysoft/FullPageOS/issues/105

4o4forbidden, 2022-11-06:
add this attribute *without parentheses ( autocomplete="off" ) to your html input !

guysoft, 2023-04-11:
Is there a solution here worth adding to the wiki?