guysoft/FullPageOS #434: High CPU usage on 2 cores

Onepamopa, 2022-07-27:
#### What were you doing?
Kiosk mode with 1 page (slider with 9 slides, 4 of which - iframes to remote pages)

[Please be as specific as possible here]

#### What did you expect to happen?
Work normally...

#### What happened instead?
High cpu usage, slider choppy

#### Was there an error message displayed? What did it say?
No

#### Version of FullPageOS?
Latest

[Can be found in /etc/fullpageos_version ALWAYS INCLUDE.]

#### Screenshot(s) showing the problem:
None

[If applicable. Always include if unsure or reporting UI issues.]

#### If you are building FullPageOS - provide a build.log that is created for the build
No

guysoft, 2022-07-27:
What web page are you using?
What version of RaspberryPi?

Onepamopa, 2022-07-27:
Pi 3 Model B (1gig ram) BCM2835 / a02082
The webpage is my doing - a slider that has some images, js + 8 iframes.
The page is restricted only for "internal use", however, I have opened it for you - https://danahotels.bg/reception-kiosk/
Let me know when I can close it.

guysoft, 2022-07-27:
Might be that something is slow in one of the panes, of the accelration is
working/not working.

You can try the non-acceleratoin variant to see if that helps. or use a Rpi
4, if its any faster it could be an indecation.

guysoft, 2022-07-27:
Might be that something is slow in one of the panes, of the acceleration is working/not working.

You can try the non-acceleration variant to see if that helps. or use a Rpi 4, if its any faster it could be an indication.

Onepamopa, 2022-07-27:
I don't have RPI4 at the moment, so cannot test with that.
As for "non-acceleration" how do I set it up ?

guysoft, 2022-07-27:
no-acceleration variant can be downloaded in the release page, flash it like you do a FullPageOS image:
https://github.com/guysoft/FullPageOS/releases/tag/0.12.0

https://unofficialpi.org/Distros/FullPageOS/no-acceleration/2021-04-14_2021-03-04-fullpageos-no-acceleration-buster-armhf-lite-0.12.0.zip