guysoft/FullPageOS #436: FullPageOS For x86

cheznewa, 2022-08-03:
#### What were you doing?

is possible to add support for x86? because raspbian existing as [x86](https://downloads.raspberrypi.org/rpd_x86/images/) as iso.

guysoft, 2022-08-03:
There are sevral issues that need to be addressed for this to happen:
1. CustomPiOS uses chroot with qemu-arm-static to build. So in order to support x86 it will need to have an option to run qemu-x86_64-static or chroot without it in a x86_64 architecture (without qemu building form arm devices would not work).
2. the x86 image that rpi provide is only the full system, there is no lite version.
3. the ISO uses squishfs inside it which would require handling of that special case of mounting the iso, creating a new squishfs (because its read only). then closing it

So at the moment I have no intention of preforming this. Of course code contributions are welcome

cheznewa, 2022-08-03:
juste you remove the package raspberrypi-ui-mods for remove the desktop
and i like tested on pc/vm without raspberry pi hardware

guysoft, 2022-08-03:
Ok, that still leaves 1 and 3

guysoft, 2022-08-03:
It might be easier to solve 3 by using armbian which is supported in CustomPiOS:
https://www.armbian.com/uefi-x86/

guysoft, 2022-09-12:
Related of duplicate: https://github.com/guysoft/FullPageOS/issues/427

guysoft, 2025-01-21:
Related: https://github.com/guysoft/FullPageOS/issues/427