guysoft/FullPageOS #439: i need a 0.7.0 version on buster

TjOsT1975, 2022-08-31:
#### What were you doing?

i need to display html sites on fullpagedashboard from local intranet

#### What did you expect to happen?

id needs to work exactly like on 0.7.0

#### What happened instead?

sites stays blank nothing is shown. with middleclick and new tab the sides are shown after proceed "your site is not pricvate"

#### Was there an error message displayed? What did it say?

blank pages on fullpageosdashboard

#### Version of FullPageOS?

0.11.0 and 0.12

[Can be found in /etc/fullpageos_version ALWAYS INCLUDE.]

#### Screenshot(s) showing the problem:

[If applicable. Always include if unsure or reporting UI issues.]

#### If you are building FullPageOS - provide a build.log that is created for the build

TjOsT1975, 2022-09-02:
Ok. 
The Log is empty.
i have a adress like this : https://user:password@jasperreports-ab-scheduler.logistics.corp/js-output/something/something.html
on fullpageos.txt with this adress inserted no problem at all.
with fullpagedashboard nothing happend. 
after open a new tab with the same adress it is working in new tab but NOT in http//localhost/fullpagedashboard/url#0
the only thing that is working using the address without user and password but this is not an option after reloading a few times it demands the user an password and nothing is happening.

I´m Shure the is a problem with username and password promt in fullpage dashboard wich not where there in 0.7.0

PLEASE i need really help.

guysoft, 2022-09-02:
You are welcome to build it yourself, you have the source. I won't support a version that came out in 2017.

closed 2022-09-02 by guysoft

TjOsT1975, 2022-09-05:
I figured out the Problem. 
The Problem is that in Dashboardview the Username und Password in the HTML Address do not work. So the screen stays blank.
It there a way to fix this?