guysoft/FullPageOS #440: Build keeps failing - seeking advise

ItsJustPW, 2022-09-09:
#### What were you doing?

I am trying to create a custom build. However, before I tried to build a custom image, I wanted to see if I could simply rebuild the default.  I created a new machine, installed the required packages (I've tried on Ubuntu and Raspbian) with same result. I'm unsure if I'm doing something wrong, or if there's a bug.

I copied [these](https://github.com/guysoft/FullPageOS#build-fullpageos-from-within-fullpageos--raspbian--debian--ubuntu) 
 steps exactly on a fresh install. 

If anyone could advise, as I'm 99% sure I'm just being stupid... that would be great.

#### What did you expect to happen?

I expected it to output an image which I would be able to use to install onto SD card as normal.

#### What happened instead?

Error when building. No information about what went wrong. 

#### Was there an error message displayed? What did it say?

BUILD FAILED!

#### Version of FullPageOS?

Latest, based on provided instructions.

#### Screenshot(s) showing the problem:

[image]

#### If you are building FullPageOS - provide a build.log that is created for the build

Log file: 
[build.log](https://github.com/guysoft/FullPageOS/files/9535888/build.log)

guysoft, 2022-09-09:
Hey,
You are not stupid, there is actually something that needs to be changed :)

The error in the log that fails the build is this:

```
'/tmp/unpack/.' -> '/patches'
'/tmp/unpack/./fix-163-autologin-console.patch' -> '/patches/fix-163-autologin-console.patch'
+ rm -r /tmp/unpack
+ pushd /usr/bin
/usr/bin /
+ patch -p1
patching file cancel-rename
Hunk #1 FAILED at 25.
1 out of 1 hunk FAILED -- saving rejects to file cancel-rename.rej
```
Its a little further up from the screenshot.

The reason its happening is that its new code that was added to handle the userless login in bullseye.

A new image came out two days ago and apparently broke that.

Related: https://github.com/guysoft/CustomPiOS/issues/163

ItsJustPW, 2022-09-09:
Thank you Guy. 

Any chance you could give me an idea of a possible solution? Sorry, I'm really new to this so I'm not understanding what to do if there is a fix currently. 

Paul

guysoft, 2022-09-09:
You could disable the userless image handeling by setting in [the config file](https://github.com/guysoft/FullPageOS/blob/devel/src/config)
``export BASE_ADD_USER=no``

That would disable the patching, but I am not sure if something else will not work since this image is new and I didn't see what was changed in it

ItsJustPW, 2022-09-09:
Thank you Guy, 

Do you have a way which I can buy you a coffee as a thank you for your help? 

Paul

guysoft, 2022-09-09:
Indeed, there is a donation button here: 
https://github.com/guysoft/FullPageOS#donate

ItsJustPW, 2022-10-11:
Thank you. I wish I could afford to contribute more but energy prices are killing me at the minute. 

To update you, it built fine (not sure where yet lol but it did) so I’m a major step ahead from where I was. 

In case youre interested. I volunteer for RNLI and I wanted something to display a web app when we get an emergency shout.  The modification I’m making is I’m going to (try) and make it log into a web page with a special account so it can display who is responding in order to better allocate crew to the particular callout/shout. 

Your work does 99% of what I wanted to do… the other part (logging in etc) I think is just a but of online research and editing. 

Thanks again for the incredible work you put into doing this. 

Paul

ItsJustPW, 2022-10-11:
> Related: guysoft/CustomPiOS#163 <https://github.com/guysoft/CustomPiOS/issues/163>

Hi Guy, 

Thank you so much for getting back to me. 
Is there anything I can do in the meantime do you know? 

Paul

guysoft, 2022-10-17:
Hey, I am not following - where are you stuck and what do you need at the currently? Last message I saw you said it builds.

zicheee, 2024-08-29:
I was running in the same issue and after checking the build.log I noticed an error that building for arm is not supported on am64.

I ended up doing the following:

1. Installing Raspberry Pi OS ([latest version](https://downloads.raspberrypi.com/raspios_arm64/images/raspios_arm64-2024-07-04/2024-07-04-raspios-bookworm-arm64.img.xz)) on a Raspberry Pi 4 (2GB)
2. Following most steps from the [tutorial](https://github.com/guysoft/FullPageOS/tree/devel?tab=readme-ov-file#build-fullpageos-from-within-fullpageos--raspbian--debian--ubuntu)
3. I downloaded the latest raspberry os (64bit): 
```shell
# in the folder: FullPageOS/src/image
wget https://downloads.raspberrypi.com/raspios_lite_arm64/images/raspios_lite_arm64-2024-07-04/2024-07-04-raspios-bookworm-arm64-lite.img.xz
```

4. Installed jq `apt-get install jq` (because build failed)
5. Added BASE_ARCH config to build for arm64 in config file:
```
nano FullPageOS/src/config
...
export USAGE_STATISTICS_URL=https://fullpageos-tracking.gnethomelinux.com

## build 64bit
export BASE_ARCH=arm64
```
7. Running `sudo bash -x ./build_dist` 

Building on my pi took about 10 minutes and I got an operating system working as expected on my Raspberry Pi 4. It took me again about 5 minutes to find the final build file :-)
It is in the workspace folder (**FullPageOS/src/workspace**)

I have not tried using the 32bit Raspberry Pi OS image on my Pi and building also a 32bit image

guysoft, 2024-08-30:
Building on arm 32bit should work. If you can provide the output from build.log it would help to debug the issue.
Happy though that you got something built

Should I close this issue since OP @ItsJustPW had a different issue.
Alternatively I think I would move this to a discussion.

zicheee, 2024-09-09:
Hi @guysoft 
I just ran into the issue, that building a 32bit image on my debian machine (VM on a Synology) was failing. After checking the tickets here, I found this issue and assumed that @ItsJustPW ran into the same issue. That's why I shared my solution which is working good so far. In the meantime I tried building "normally" on a fresh raspberry pi using the official 32bit [Raspberry Pi OS](https://downloads.raspberrypi.com/raspios_lite_armhf/images/raspios_lite_armhf-2024-07-04/2024-07-04-raspios-bookworm-armhf-lite.img.xz?_gl=1*1uhdyhx*_ga*MTA2OTE2NDI4MC4xNzI1ODg3NTY1*_ga_22FD70LWDS*MTcyNTg4NzU2NS4xLjEuMTcyNTg4NzU2OS4wLjAuMA..) and it was working without modifying the config file and chaning the BASE_ARCH. 

But in both cases: When using the Raspberry Pi OS lite, **jq** must be installed manually as this software is not part of the image. 

I cannot give you the output of my build.log, as I already deleted the VM. But the error was very clear and mentioned the architecture. That's why now I am using a pi to build the OS for my PI. My home assistent touch panel is now working like charm with the os.

guysoft, 2024-09-09:
Hey @zicheee , can you provide a build.log ?

Also jq is needed to be installed because its part of CustomPiOS dependencies:
https://github.com/guysoft/CustomPiOS?tab=readme-ov-file#requirements