guysoft/FullPageOS #441: Resolution seemingly not working

IfgiU, 2022-10-31:
#### What were you doing?

Hi! I was trying to set the resolution to 1920x1080 in `/boot/config.txt` by adding these values:
`hdmi_group=1
hdmi_mode=16`

#### What did you expect to happen?

I expected that to set the resolution to 1920x1080.

#### What happened instead?

Nothing, these values were completely ignored. I also tried setting the resolution using `raspi-config`,
but it just did what I did and also changed `hdmi_force_hotplug` to 1, resulting in the resolution and aspect ratio to be completely messed up.

#### Was there an error message displayed? What did it say?

No.

#### Version of FullPageOS?

0.12.0, tho I did some upgrades using apt update und upgrade.

####

IfgiU, 2022-10-31:
I also tried setting group to 2 and mode to 82. Same thing.

brynyneuadd, 2023-01-01:
Yes, I've got this problem too - have a rpi4 attached to the same monitor using a different raspbian set up and it works perfectly, so suspect a problem in fullpageOS somewhere.... tried dozens of different configurations, but just defaulting back to standard resolutions.

Curiously it appears to boot in 4k on my TV and can see the raspbian boot text in 4k, then when it goes into fullbootOS, it lowers the resolution...

guysoft, 2023-01-26:
Does this happen also on RaspberrypiOS? Or is only in FullPageOS?

shaunhurley, 2023-03-02:
Folks, dealing with the same (similar) issue, found a few details and the (a?) solution that works manually for me, however need to figure out where to put it so it happens on startup every time.

In my case, the physical display / VNC sessions are displaying 3840 x 2160 whereas my kiosk webpage output is fixed at 1920 x 1080, so I need to force 1920 x 1080 resolution.

[image]

The problem is that X11 on pi4 does not use/honor the HDMI group/mode settings from /boot/config.txt.  From [this thread](https://forums.raspberrypi.com/viewtopic.php?t=258612):

> hdmi_mode and hdmi_group are passed to the kernel, so if you boot to a console then they will apply.
> X11 (the desktop) ignores the request made to the kernel. It takes the list of available modes (as parsed from the monitor EDID) and chooses a default. You can change the mode through the Screen Configuration tool, and that change should be persistent across reboots.
> 
> The one exception is if you use hdmi_group=2, hdmi_mode=87, hdmi_timings=... to set up a custom timing. If you do that then the EDID is NOT parsed, and your one custom mode is passed through KMS.

The solution (if you don't want to get into custom timings, etc. :) is to run [/usr/bin/xrandr](https://www.x.org/releases/X11R7.5/doc/man/man1/xrandr.1.html) and specify the desired resolution. e.g. for 1920 x 1080:

```
/usr/bin/xrandr` --output HDMI-1 --mode 1920x1080
```
NOTE:  to execute that command remotely via an SSH shell you have to prefix it with [DISPLAY=:0](https://unix.stackexchange.com/questions/193827/what-is-display-0) like:
```
DISPLAY=:0 /usr/bin/xrandr` --output HDMI-1 --mode 1920x1080
```
After execution this switches the (my) 4K TV output to 1920x1080 on the physical TV, and appears sized correctly when I VNC to the device.

Next step is to figure out where\how to have this execute on startup...

shaunhurley, 2023-03-02:
Found how to configure X11 on pi4 at startup to achieve same results (force a 4K display into HD resolution), without trying to embed an xrandr call somewhere, etc.

1. SSH to the Raspberry Pi
2. Change to the X11 configuration directory
```
cd /usr/share/X11/xorg.conf.d
```
3. Create a file called “monitors.conf” and open it for editing
```
sudo touch monitors.conf
sudo nano monitors.conf
```
4. Configure a section for the primary HDMI output and save the file
```
Section "Monitor"
  Identifier "HDMI-1"
  Option "Primary" "false"
EndSection
```
5. Create a file called “screen-resolution.conf” and open it for editing
```
sudo touch screen-resolution.conf
sudo nano screen-resolution.conf
```
6. Configure the desired screen resolution for the HDMI monitor and save the file
```
Section "Screen"
  Identifier "HDMI-1"
  SubSection "Display"
    Modes "1920x1080"
  EndSubSection
EndSection
```
7. Reboot the device
```
sudo shutdown -r now
```

All things being equal, your display should now be in 1920 x 1080 mode.

guysoft, 2023-11-07:
@shaunhurley thats cool, if you want you can add that to the FAQ in the wiki:
https://github.com/guysoft/FullPageOS/wiki/FAQ

shaunhurley, 2023-11-08:
Done! :)

guysoft, 2023-11-08:
Thanks for contributing!