guysoft/FullPageOS #442: Cannot access FullPageDashboard from another computer on the same network.

AirWind001, 2022-11-03:
I tried to access the fullpagedashboard from my laptop which is connected to the same wifi network that the RPi is using. i tried both addresses http://fullpageos.local/FullPageDashboard and http://<ip address>/FullPageDashboard but it says " localhost connection refused
[image]

Wheres on RPi when viewed from VNC the dashboard works fine
[image]

#### Version of FullPageOS
pi@fullpageos: $ cat /etc/fullpageos_version
0.12.0

4o4forbidden, 2022-11-06:
Hello, find the local address of the pi ( ex : 192.168.1.6 )  and try to go like this : 
http://192.168.1.6/FullPageDashboard

AirWind001, 2022-11-07:
Hi, thanks for the reply. 

That's what I've done actually, you can see in the first screenshot I've entered http://192.168.178.150/FullPageDashboard. You can also see that part of the UI loads but the content of each tab didn't load.
[image]

guysoft, 2022-11-07:
@AirWind001 Can you also provide the console output?
For chrome: Option + ⌘ + J (on macOS), or Shift + CTRL + J (on Windows/Linux).
For Firefox: Ctrl + Shift + J (or Cmd + Shift + J on a Mac

There is likely some kind of cross site policy of something like that.

AirWind001, 2022-11-07:
Thank you for the reply @guysoft 
below is the screenshot. when i hover my mouse to the center of the webpage it says "localhost refused to connect"
[image]
This is on my home network, and it shows the same result as before when i tried it on the office network. 
Also interesting to note ( if i didn't mention it already) that i can access other pages successfully (http://192.168.0.105/imagetest/) only the FullPageDashboard page has this issue
[image]