guysoft/FullPageOS #443: Black Screen / Fix This Please

4o4forbidden, 2022-11-07:
#### What were you doing?

Normal Boot

#### What did you expect to happen?

Boot to initial page

#### What happened instead?

Black screen but the pi connects to wlan and i can access it via SSH

#### Was there an error message displayed? What did it say?

None

#### Version of FullPageOS?

I tested both nightly and official

[Can be found in /etc/fullpageos_version ALWAYS INCLUDE.]

#### Screenshot(s) showing the problem:

[If applicable. Always include if unsure or reporting UI issues.]

#### If you are building FullPageOS - provide a build.log that is created for the build

cengbrecht, 2022-11-19:
Being a brand new user of it, I noticed that it puts up a warning that it cannot access home/pi
Did you happen to rename your primary user?

4o4forbidden, 2022-11-19:
No ! it's always pi

cengbrecht, 2022-11-20:
> No ! it's always pi

Ok, sorry for the inconveience, I noticed that a lot of the config files are hard coded for pi, so I was making sure you didn't do what I did. 👍🏼

4o4forbidden, 2022-12-09:
@guysoft  any help please ?

guysoft, 2023-01-15:
What type of screen are you using?
I am not sure how to debug this, it looks like the screen might need  a different setting - did you try another screen?