guysoft/FullPageOS #448: Screen blanking on current version, was not on long running 0.8.1 instances

Teku16, 2022-12-15:
#### What were you doing?

Normal operation of FullPageOS - Upgraded from a 2017 version (0.8.1?) finally, as we were having odd PDF encoding issues with viewing documents inside iFrames -- I think this was a chromium issue. 

Upgrading to the current release fixed that, but we are now experiencing black screening, at random intervals, across all 4 instances we are running. I have tried using our old (modified) start_chromium script and the stock one with the correct URL, with no luck. I have tried combinations of the screensaver/blanking settings, both accel and no-accel versions, and I am still experiencing the random blanking anywhere from 8-36 hours after reboot. After the blacking, I can still SSH into these. Nothing about the page I am loading changed. These are touchscreen monitors made by Viewsonic, I believe. 

I am at my wits end with these and about to try a previous release, as I read something about buster/bullseye causing the issue among certain monitors, but I would prefer to stay on the newest release as it is much faster with my css animations. 

#### What did you expect to happen?

Pi stays on indefinitely as in previous releases

#### What happened instead?

Pi black screens after anywhere between 8-36 hours 

#### Was there an error message displayed? What did it say?

Unsure where to look

#### Version of FullPageOS?

Current

##### our (old) start_chromium script which grabs device ID from a single line text file
```#!/bin/bash
file='/boot/device_id'
read id < $file
chromium-browser --disable-infobars --touch-events=enabled --disable-pinch --noerrdialogs --disable-session-crashed-bubble --incognito <url>?device=$id```

Teku16, 2023-01-03:
Happy New Year! Any thoughts on this?

guysoft, 2023-01-03:
What website are you using? 
Does this happen on any website or i the one you are using in particular?

Teku16, 2023-01-04:
> What website are you using? Does this happen on any website or i the one you are using in particular?

These devices are offline on a local network, loading custom-built html/php pages as part of a digital information/recordkeeping system. Unfortunately I cannot link to anything for you, but I can answer further questions probably

Teku16, 2023-01-04:
Some more info. I discovered that these pi's are crontab'd to reboot nightly at 1am, Therefore, the random blanking is happening randomly anywhere from 10-20hrs in, and in some cases, not at all between reboots. We had one stay live 24/7 thru Christmas and New Years, only to dark on us today around 10am... I downgraded one to `2019-04-08-fullpageos-no-acceleration-stretch-lite-0.10.0` and it has made it 24h with no incidents. 

I also tried all of the suggested fixes to waking the screen on this thread: https://raspberrypi.stackexchange.com/questions/10374/wake-console-screen-with-ssh

guysoft, 2023-01-14:
If I may ask  - where are you running these?
It might be something related to the type of screen.

Perhaps looking at ``journalctl -b -1`` output might show if something happened in the kernel output of the previous boot?
Of if you can connect to one that is blank (can you?) look at the dmesg output