guysoft/FullPageOS #450: Webpage Might be Temporarily Down or it may Have Moved (websites with self signed certificate)

zaheerabbas1988, 2023-01-19:
#### What were you doing?
Trying to display a simple locally hosted grafana dashboard

[Please be as specific as possible here]
Having issues displaying websites with self-signed certificates (hosted locally on LAN i.e., zabbix, grafana). Getting standard chrome error i.e., Webpage Might be Temporarily Down or it may Have Moved. Can ping and telnet, so there isnt any firewall issue. 
Can display all other public or local pages without any issues. 

#### Was there an error message displayed? What did it say?
Looked into debug and Syslog, but could not find anything. 

#### Version of FullPageOS?
2021-04-14_2021-03-04-fullpageos-buster-armhf-lite-0.12.0

guysoft, 2023-01-26:
Its an issue that browser makers are not taking self-signed services in to account.
I also have in OctoPrint and locally hosted https.

I did not try this, but you should be able to add the certificate to the FullPageOS system and then it should be recognized.
Here is a guide how to do that:
https://support.kaspersky.com/ScanEngine/1.0/en-US/182984.htm

zaheerabbas1988, 2023-01-26:
Thank you, however, my issue needed an upgrade to the chromium browser. After that, I was able to see the standard accept self-signed certificate message.