guysoft/FullPageOS #451: RPi alternatives?

MKaidan, 2023-01-26:
What would be an alternative to running this on an RPi?

guysoft, 2023-01-29:
The image at the moment is build only for RaspberryPi. I was in talks with @libre-computer-project to see if we could also support their boards, but at the moment I have no boards to test with.

SLC-Josh, 2023-02-01:
I will test on my libre computer "le potato" tonight. I don't have a compatible touch screen but will confirm other functions work as expected. Their portability tool is actually pretty good.

SLC-Josh, 2023-02-03:
Okay, so I have confirmed functionality with Libre's "le potato" using the portability tool (Used a mouse and screen since I don't have a compatible touchscreen at the moment). If I can figure out how to use the CustomPiOS I will try to build an Armbian version for the BananaPi zero that I have.

guysoft, 2023-02-05:
Cool. this tool right?
https://github.com/libre-computer-project/libretech-raspbian-portability

SLC-Josh, 2023-02-06:
That's the one. It auto coverts most Raspbian images (Things using NOOB don't work). From what I've looked into it, it seems to update the kernel and a few other files needed to boot. I wish all pi alternatives had this tool.

dtseiler, 2023-03-14:
Would love to see this ported to Libre as well. Can we donate to get a board in your hands?

I bought a couple of Libre Renegades but so far every RPi project I had hoped to use them with seems to have RPi-specific software libraries that require an actual RPi.

guysoft, 2023-05-18:
I am looking in to adding support for @librecomputer since they sent me hardware to test. Its in progress.

geeklimit, 2023-06-12:
How had the testing been going? Is there anything we can do to help?

guysoft, 2023-06-12:
I've started writing a multi device build for OctoPi.
When that works I can add that to FullPageOS. Thanks for askin because that lets me know its needed here too.

Progress is at:
https://github.com/guysoft/OctoPi/tree/ci-orange/config

geeklimit, 2023-06-13:
Great to see - I wouldn't be surprised to see an uptick in the projects, due just to the availability of hardware!

geeklimit, 2023-07-11:
@guysoft it seems like this is ready for testing, based on what I see here: https://github.com/guysoft/CustomPiOS

Or a I reading that wrong?

guysoft, 2023-07-17:
Its not fullu ready yet. However if you are ready to test i can try make time for it

geeklimit, 2023-07-17:
I'm happy to help in any way I can! I've got a half-dozen le potatoes ready for the baking

guysoft, 2023-07-19:
Ill see if i can get that going

geeklimit, 2023-10-03:
Anything I can help test? I still have a pile of Le Potato devices available.

dtseiler, 2023-10-03:
> Anything I can help test? I still have a pile of Le Potato devices available.

Same here, I have Le Potatoes and Renegades that I can re-purpose to test.

guysoft, 2024-01-15:
Related, this is effort to add support for this:
https://github.com/guysoft/CustomPiOS/issues/214

SecularSteve, 2024-02-07:
We have a Discussion Board now. In case you see this.

guysoft, 2024-02-07:
I could convert this to a discussion, but not sure I should. Depends what @MKaidan prefers.

guysoft, 2024-03-31:
Hey, there is progress. Its taking longer because it became part of CustomPiOS v2. Which I've been working on:
https://github.com/guysoft/CustomPiOS/issues/214

I can make a variant for LibreComputer but didn't get an answer to which base image to use:
https://github.com/guysoft/CustomPiOS/issues/214#issuecomment-1951057150

@Yukigamine What portability script are you using? If I can take a look at that I can get bullseye support still going.
Also if you can figure out what base image to use that would help me greatly getting it to work, perhaps even now in the CustomPiOS v2 branch that is here:
https://github.com/guysoft/CustomPiOS/tree/feature/meta-modules

guysoft, 2024-11-02:
Hey, building on lepotato Debian:
```
Reading state information...
Package chromium-browser is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

W: --force-yes is deprecated, use one of the options starting with --allow instead.
E: Package 'chromium-browser' has no installation candidate
```
Need to figure out if there is a chromium package, or move images.