guysoft/FullPageOS #452: Add jq to the apt install line in the readme for building?

OmgItsBkid, 2023-02-07:
I know it's minor, but it seems that neither Raspberry Pi OS nor Ubuntu come with jq by default, so I thought it might be worth a mention. It's on the CustomPiOS readme, but not here.

#### What were you doing?

Building in Ubuntu 22.04.1

#### What did you expect to happen?

The build to succeed.

#### What happened instead?

The build failed. ( jq: command not found)

#### Was there an error message displayed? What did it say?

+++ echo 'Adding 1000 MB to partition 2 of 2022-09-22-raspios-bullseye-armhf-lite.img'
Adding 1000 MB to partition 2 of 2022-09-22-raspios-bullseye-armhf-lite.img
++++ sfdisk --json 2022-09-22-raspios-bullseye-armhf-lite.img
++++ jq '.partitiontable.partitions[] | select(.node ==  "2022-09-22-raspios-bullseye-armhf-lite.img2").start'
/home/user/FPOS/CustomPiOS/src/common.sh: line 280: jq: command not found
+++ start=

#### Version of FullPageOS?

N/A

#### Screenshot(s) showing the problem:

N/A

#### If you are building FullPageOS - provide a build.log that is created for the build

Attached.
[build.log](https://github.com/guysoft/FullPageOS/files/10679529/build.log)

guysoft, 2023-02-07:
Valid point, its a requirement of CustomPiOS, that is why its failing:
https://github.com/guysoft/CustomPiOS#developing

I could add it, and to  all the other images I maintain. Usually you don't list dependencies of a dependency. But since it got you it might be worth noting.

OmgItsBkid, 2023-02-08:
That makes a lot of sense, and until I did some research I didn't realize it was listed under CustomPiOS. After getting the error, I just grabbed the package and continued on my way. I really only mentioned it because it might be a trip up in an otherwise smooth process.

commit 0013dea5 (2023-02-08): Add Requirements for smoother build process fixes #452
closed 2023-02-08 by guysoft