guysoft/FullPageOS #453: refused to connect Internal Grafana with Lets Encrypt SSL Cert

matze1708, 2023-02-08:
#### What were you doing?

I would like to call an internal page to our Grafana. 

#### What did you expect to happen?

Show Grafana Site

#### What happened instead?

I get an Error refused to connect

#### Was there an error message displayed? What did it say?

grafana.intranet.XXXXXX.de refused to connect.
Site is only availablefrom our intranet.

#### Version of FullPageOS?
0.12.0
[Can be found in /etc/fullpageos_version ALWAYS INCLUDE.]

guysoft, 2023-02-09:
Does it connect if you open a browser window using ctrl+T from FullPageOS?

matze1708, 2023-02-09:
Hello thanks for the answer
Yes, it works through this browser / access

it only loads very long, but the dashboard is also large

chadmccune, 2023-02-22:
I'm getting the same issue, but its with every url when trying to use the dashboard to rotate.  nothing works, even https://www.google.com

guysoft, 2023-02-23:
It might be cross-site https request.

anduz, 2024-05-14:
Hi, I'm having this problem as well.
Also I would need to automatically login to a web-site, is that possible?

anduz, 2024-05-21:
It is NOT working. Afert some time I get the error again.
Rebooting does not solve the problem, only after when I open another tab, the fullpage tab works after refreshing

guysoft, 2024-05-26:
@anduz Are you using grafana too? What userscript?

Also I think I am converting this issue in to a discussion, it does not seem to be a bug, just a specific usecase (one that a few years ago I did too)