guysoft/FullPageOS #456: Installation from debian scratch

Stephim, 2023-03-16:
#### What were you doing?

[Please be as specific as possible here]

#### What did you expect to happen?

#### What happened instead?

#### Was there an error message displayed? What did it say?

#### Version of FullPageOS?

[Can be found in /etc/fullpageos_version ALWAYS INCLUDE.]

#### Screenshot(s) showing the problem:

[If applicable. Always include if unsure or reporting UI issues.]

#### If you are building FullPageOS - provide a build.log that is created for the build

guysoft, 2023-03-16:
You will need to build it using a debian/armbian base image. This repo hold scripts to do that

roblesterjr04, 2023-03-20:
I've attempted this script a couple times. What exactly is the process? It spits out a new img file in the workspace directory, and I'm able to flash it to a card using the RPI imager. But when it boots its like it was untouched. just a vanilla installation.

guysoft, 2023-04-03:
Can you provide the generated build.log ? Otherwise I can't see what it is doing.