guysoft/FullPageOS #459: Every other safe_refresh call I make in crontab leads to a blank page with the Chrome Web Store icon

MKaidan, 2023-03-27:
#### What were you doing?

Using crontab to safe_refresh the page every day (twice a day).

#### What did you expect to happen?

The page would refresh successfully.

#### What happened instead?

The page goes to a blank screen with the Chrome Web Store icon in the middle.

#### Was there an error message displayed? What did it say?

No.

#### Version of FullPageOS?

0.13.0

#### Screenshot(s) showing the problem:

[image]

#### If you are building FullPageOS - provide a build.log that is created for the build

MKaidan, 2023-03-28:
After a reboot, now I get this error when I perform `safe_refresh`:
```
Refreshing page.
Invalid MIT-MAGIC-COOKIE-1 keyError: Can't open display: (null)
Failed creating new xdo instance
Invalid MIT-MAGIC-COOKIE-1 keyError: Can't open display: (null)
Failed creating new xdo instance
Invalid MIT-MAGIC-COOKIE-1 keyError: Can't open display: (null)
Failed creating new xdo instance
Invalid MIT-MAGIC-COOKIE-1 keyError: Can't open display: (null)
Failed creating new xdo instance
```

MKaidan, 2023-03-28:
After a second reboot, things are back to normal, but what could be causing these issues?

guysoft, 2023-04-02:
Looks there is something wrong with your display - how are you connecting to the Pi? I might be that DISPLAY environment variable is not set correctly, or you are using SSH with X forwarding:
https://unix.stackexchange.com/questions/199891/invalid-mit-magic-cookie-1-key-when-trying-to-run-program-remotely

closed 2023-04-02 by guysoft

MKaidan, 2023-04-03:
Why would any of these be the issue with a standard FullPageOS install?

guysoft, 2023-04-03:
It depends how you are connected to it. Over ssh, a virtual terminal or terminal inside the graphical interface. These would yield different results

guysoft, 2023-05-18:
Who is "We"?

This needs to be set correctly:
https://datacadamia.com/ssh/x11/display